dsmrreader/dsmr-reader #946: DSMR data in home assistant afbeelden

heystge, 2020-03-30:
Ik heb Hass.io draaien op de ene Raspberry Pi en DSRM-reader op een andere Pi.
DSMR laat zijn eigen data goed zien in het dashboard.
Ik heb de MQTT functie aangezet en het lijkt erop de dat de connectie ook werkt.
In Home Assistant zie ik in de logging ook data binnenkomen.....

bijv:
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity2
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity1_returned
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity2_returned
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity_merged
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity_returned_merged

Ik heb echter nu al alle mogelijke configuraties geprobeerd in de YAML file, echter op geen enkele wijze krijg ik data te zien.
sensor:

platform: dsmr_reader
maakt netjes alle sensoren aan, deze zie ik ook bij 'toestanden' in de lijst staan, echter allemaal met 'unknown' als binnenkomende data:
sensor.dsmr_day_consumption_electricity2 | unknown | unit_of_measurement: kWh friendly_name: High tariff usage icon: mdi:counter

Welke essentiele stap lijk ik nu te missen om de data netjes op een LoveLace pagina te kunnen afbeelden?

dennissiemensma, 2020-03-30:
Bedankt voor je vragen. Ik kan je helaas hier niet helpen, omdat ik er zelf geen ervaring mee heb. Ik gebruik zelf zowel geen HA als LoveLace.

Wat ik wel kan doen, is je richting techneuten sturen die hier veel en veel meer verstand van hebben en die jou waarschijnlijk ook wel wat tips willen geven. 
Zie hiervoor o.a. dit topic op Tweakers: https://gathering.tweakers.net/forum/list_message/61927308

Volgens mij komen daar diverse dingen aan bod met HA en andere combinaties. Je zou het daar kunnen voorleggen en anders willen ze je vast doorsturen naar het juiste topic met meer info.

closed 2020-03-30 by dennissiemensma

heystge, 2020-03-31:
Dank je wel!!