dsmrreader/dsmr-reader #947: Live grafiek langer dan 1 dag

Phyxion, 2020-03-31:
Misschien mis ik ergens een instelling, maar ik zou graag de live grafiek langer dan 1 dag willen tonen (minimaal 2 dagen). Ik heb nu al regelmatig gehad dat de grafiek net buiten de interessante periode valt om te kunnen vergelijken.

dennissiemensma, 2020-03-31:
Je kunt dit gedeeltelijk instellen in de frontend/weergave instellingen in de admin, onder `Dashboard graph width`. Dit geeft het aantal meetpunten aan dat standaard wordt getoond in de grafiek, maar werkt alleen voor een deel van de grafieken.

Ik vermoed dat je dit wellicht juist in de grafieken wilt zien (gas, temperatuur) waarbij bovenstaande instelling geen effect heeft. Klopt dat?

Phyxion, 2020-03-31:
Dat had ik al geprobeerd maar dat past alleen de standaard zoom van de grafieken aan, de gegevens blijven niet verder teruggaan dan één dag (ook op bijvoorbeeld 9999 niet). De reden dat ik meer dan één dag wil zien is omdat je vaak bijvoorbeeld even de laatste paar uren wilt bekijken, maar die kan je dan niet meer vergelijken met de dag daarvoor omdat die gegevens al uit het overzicht verdwenen zijn. Een instelbare tijd lijkt mij ideaal, maar als het makkelijker is om het eerst op 2 of 3 dagen te zetten dan is dat ook welkom 👍🏻

dennissiemensma, 2020-03-31:
Prima. Lijkt me dan een goede toevoeging voor een toekomstige release.

dennissiemensma, 2020-03-31:
Extra suggesties vanaf #926

dennissiemensma, 2020-05-03:
Voor de volgende release kun je dit instellen in aantal uren (1 tot 168). Daarnaast zijn alle live grafieken standaard uitgezoomed.

Het verhogen van het aantal zichtbare uren kan invloed hebben op de laadtijd of performance in de browser.

closed 2020-05-03 by dennissiemensma

Phyxion, 2020-05-04:
Super, dat het wat langer duurt is geen probleem.

Mixpower, 2020-05-05:
> Voor de volgende release kun je dit instellen in aantal uren (1 tot 168). Daarnaast zijn alle live grafieken standaard uitgezoomed.
> 
> Het verhogen van het aantal zichtbare uren kan invloed hebben op de laadtijd of performance in de browser.

Hoe kan ik dit weer terugzetten, ik wil de live grafieken namelijk niet uitgezoomed.

Phyxion, 2020-05-05:
Werkt helemaal prima zo, bedankt @dennissiemensma Laadtijd valt reuze mee.

dennissiemensma, 2020-05-05:
@Mixpower ik heb in 3.9.1 een optie toegevoegd om zelf de zoom (voor de grafieken met veel data) te kunnen bepalen. Voor nu zou dat een redelijk tussenweg moeten zijn.

Mixpower, 2020-05-05:
> @Mixpower ik heb in 3.9.1 een optie toegevoegd om zelf de zoom (voor de grafieken met veel data) te kunnen bepalen. Voor nu zou dat een redelijk tussenweg moeten zijn.

Ah super, dat werkt inderdaad een stuk beter! :)