dsmrreader/dsmr-reader #948: Telegrams via API worden niet verwerkt bij teruglevering

jghaanstra, 2020-03-31:
**Wat gebruik je?**
* DSMR-reader versie: v3.7
* Hardware: Synology NAS DS216+II
* Native of Docker: Docker container van xirixiz
* DSMR-protocol slimme meter *(v4 of v5)*: v4.2

**Omschrijving**
Ik probeer DSMR Reader te gebruiken middels de API nadat ik problemen had met het inlezen van telegrams (telegram stopten met binnenkomen na 1 - 3 dagen). Dit leek prima te werken maar ik loop nu tegen een vermoedelijke bug aan waarbij telegrams niet worden verwerkt als er meer energie wordt opgewekt dan wordt verbruikt. Ik kom tot deze conclusie nadat vandaag het inlezen stopte precies nadat ik ging terugleveren en weer werd hervat nadat er weer meer werd verbruikt en ik de onverwerkte telegrams uit de database heb verwijderd. Nou zou het kunnen dat er iets mis is met de telegram berichten die ik aanlever via de API op het moment dat ik meer energie opwek. Maar ik vermoed dat het een probleem is met DSMR Reader aangezien er niks anders is aan het telegram bericht en ik ze ook zie binnenkomen (ze worden alleen niet verwerkt).

Hieronder staat de payload van twee API calls, de eerste is goed verwerkt maar de tweede wordt niet verwerkt. Zou je hier eens naar willen kijken.

```'{"timestamp": "2020-03-30 19:15:28.833721+00:00","electricity_currently_delivered":0.349,"electricity_currently_returned":0.0,"electricity_delivered_1":2591.137,"electricity_delivered_2":2359.16,"electricity_returned_1":1595.7,"electricity_returned_2":2997.061,"phase_currently_delivered_l1":0.347,"phase_currently_returned_l1":0.0,"extra_device_timestamp":2020-03-30 19:13:08.513725+00:00,"extra_device_delivered":3399.493}'`
```

```
'{"timestamp": "2020-03-31 09:48:52.243199+00:00","electricity_currently_delivered":0.0,"electricity_currently_returned":2.078,"electricity_delivered_1":2592.366,"electricity_delivered_2":2361.996,"electricity_returned_1":1595.7,"electricity_returned_2":3001.966,"phase_currently_delivered_l1":0.0,"phase_currently_returned_l1":2.077,"extra_device_timestamp":"2020-03-31 13:02:51.245231+00:00","extra_device_delivered":3402.3}'
```

dennissiemensma, 2020-03-31:
Bedankt voor je melding. Ik denk dat je het beste debug-logging kunt aanzetten en dan in de log van dsmr_backend kijken wat het systeem doet en of die ergens op hangt of een foutmelding geeft: https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html#logging

Gezien je Docker gebruikt zul je even moet uitvogelen hoe je dit precies in de Docker-container doet. Als het werkt dan hoor ik graag of je fouten of iets anders geks ziet in bovenstaande log.

jghaanstra, 2020-04-01:
Debug logging aanzetten conform de documentatie van DSMR Reader lijkt helaas niet mogelijk te zijn in de Docker container aangezien supervisor niet beschikbaar is en `./post-deploy.sh` niet werkt. @xirixiz , zou je dit kunnen bevestigen?

Nog wat aanvullende constatering. De telegrams die worden aangeboden aan DSMR Reader waarbij export groter is dan import worden geparkeerd totdat import weer groter is dan export. Daarna worden deze telegrams gedropt en worden de telegrams waarbij import weer groter is verwerkt. Echter gedurende de periode dat dit gebeurd zie ik de live statistiek van import (teruggeleverd) wel bijgewerkt worden met de actuele waarde. De telegrams worden dus wel degelijk gelezen maar niet verwerkt in de statistieken.

Misschien als ik ooit meer tijd heb koppel in een RPi aan de meter om dit te reproduceren op een omgeving waar debug logging wel beschikbaar is maar dat zal waarschijnlijk niet binnenkort zijn.

xirixiz, 2020-04-01:
Hi @jghaanstra,

For debugging you can set the following env var:
```
- DSMRREADER_LOGLEVEL=WARNING # Valid values are WARNING, INFO, DEBUG
``

With the latest image, during startup, you'll be notified about the path where the logfiles can be found:
```
docker logs -f <name of the container>
```

Je kan ook "inloggen" op de container en dan de logfiles bekijken. Ik heb het geverifieerd en wanneer het loglevel is aangepast, dan worden de dsmr gerelateerde logfiles ook netjes op dat loglevel gezet:
```
docker exec -ti <name of the container> bash
cd /var/log/supervisor
```
Daar staan dan alle dsmr reader gerelateerde logfiles, dus niet de logfiles van supervisord (de wrapper om alle dsmr reader processen heen).

dennissiemensma, 2020-04-01:
@jghaanstra ik heb meer informatie nodig over wat je exact waar ziet.

- Wat geeft de API terug op je telegrammen, worden die wel geaccepteerd?
- Standaard worden telegrammen gegroepeerd verwerkt per minuut, kan het zijn dat je daar tegenaan loopt?
- Waar zie je dat een telegram wel of niet verwerkt wordt, kijk je rechtstreekt in de database?

jghaanstra, 2020-04-01:
@xirixiz bedankt voor de aanvullende informatie. Het lukt daarmee inderdaad om debug logging aan te zetten. Een eerste blik op de logging laat helaas geen bijzondere meldingen zien.

@dennissiemensma om je vragen te beantwoorden.

- De POST API calls worden gewoon met een 201 beantwoorden en komen in de queue van onverwerkte telegrams te staan in DSMR. Ik heb de debug logging vergeleken tussen een call waarin import hoger is dan export en visa versa maar zie hier geen verschil in. Hieronder een stukje van deze logging.
- Nee, als de zon flink schijnt kan het zo een paar uur duren voordat er iets verwerkt wordt (b.v. als de waterkoker wordt aangezet en de import weer even hoger is dan de export.
- Ik kijk onder status naar het aantal onverwerkte telegrams. In de database hebben deze de waarde processed = false. Deze entries lijken te worden gedropt nadat verwerking van nieuwere telegram berichten waarbij import hoger is dan export wordt hervat maar dit zou ik even moeten verifieren.

Aangezien een plaatje meer zegt dan een tekst heb ik een screenrecording gemaakt. Bij de start is de import nog net hoger dan de export (einde van de dag en zonnepanelen leveren niet meer veel op). Je ziet ook dat export wel veranderd. Onder status zie je een aantal onverwerkte telegram berichten (start met 16 maar loopt op tot 21). Daarna zet ik de waterkoker aan en is import hoger dan export. Telegrams worden verwerkt en zakt uiteindelijk weer (in dit filmpje naar 13 maar uiteindelijk naar de standaard 2 a 3). Hopelijk is het hiermee wat duidelijker.

https://drive.google.com/open?id=1KFCDwvNXaPI6Yy12XZ7NRUqgKibGgU73

Hieronder nog een stuk logging waar ik zelf niks raars in zie.

```

2020-04-01 17:53:39,547 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:39,546] DEBUG    SP: 0 backend service(s) ready to run
2020-04-01 17:53:39,577 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:39,577] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2020-04-01 17:53:39,682 DEBG 'nginx' stdout output:
192.168.0.110 - - [01/Apr/2020:17:53:39 +0200] "POST /api/v2/datalogger/dsmrreading HTTP/1.1" 201 766 "-" "HomeAssistant/0.107.7 aiohttp/3.6.1 Python/3.7" "-"

2020-04-01 17:53:40,010 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:40 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"
2020-04-01 17:53:40,588 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:40,587] DEBUG    SP: 0 backend service(s) ready to run

2020-04-01 17:53:40,606 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:40,606] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

2020-04-01 17:53:41,102 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:41 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"
2020-04-01 17:53:41,616 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:41,615] DEBUG    SP: 0 backend service(s) ready to run
2020-04-01 17:53:41,638 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:41,638] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

2020-04-01 17:53:42,007 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:42 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"

2020-04-01 17:53:42,648 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:42,647] DEBUG    SP: 0 backend service(s) ready to run

2020-04-01 17:53:42,666 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:42,665] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

2020-04-01 17:53:43,011 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:43 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"

2020-04-01 17:53:43,676 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:43,675] DEBUG    SP: 0 backend service(s) ready to run
2020-04-01 17:53:43,694 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:43,694] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

2020-04-01 17:53:44,010 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:44 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"

2020-04-01 17:53:44,704 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:44,703] DEBUG    SP: 0 backend service(s) ready to run

2020-04-01 17:53:44,724 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:44,724] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2020-04-01 17:53:45,401 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:45 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"
2020-04-01 17:53:45,732 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:45,731] DEBUG    SP: 1 backend service(s) ready to run
[... cut ...]
[2020-04-01 17:53:50,043] DEBUG    SP: 0 backend service(s) ready to run
2020-04-01 17:53:50,067 DEBG 'dsmr_backend' stdout output:
[2020-04-01 17:53:50,067] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2020-04-01 17:53:50,403 DEBG 'nginx' stdout output:
172.17.0.1 - - [01/Apr/2020:17:53:50 +0200] "GET /xhr/header HTTP/1.1" 200 141 "http://192.168.0.110:7777/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.149 Safari/537.36" "-"
```

dennissiemensma, 2020-04-01:
Bedankt voor de uitgebreide toelichting. Er zijn diverse dingen die hier in kunnen meespelen, maar ik zie zelf niet zo 1-2-3 een verband tussen de inhoud van de gegevens en het wel of niet verwerken. Het enige wat daar invloed op heeft zijn de systeemtijden van de slimme meter en het OS van DSMR-reader.

Je geef trouwens aan dat je een DSMR-v4 meter hebt. Klopt het dat je elke 10 seconden dan een meting hebt? Ik zie namelijk in je filmpje dat je voor 17:52 zo'n `20` metingen hebt, terwijl je er max `6` a `7` per minuut zou moeten binnenkrijgen (mits de interval van 10 seconden dus klopt).

----

Verder, zou je eens direct in de database kunnen kijken? Ik ben dus niet thuis in docker, maar wellicht werkt dit al voor je:

```
docker exec -t dsmrdb psql dsmrreader -U dsmrreader
```

Je kunt dan een van deze queries een paar keer uitvoeren in een minuut, om het verloop te zien:
```
# Toon alle onverwerkte metingen
select id, timestamp from dsmr_datalogger_dsmrreading where processed = false order by timestamp desc;
```

```
# Toon laatste 20 metingen + status
select id, timestamp, processed from dsmr_datalogger_dsmrreading order by timestamp desc limit 20;
```

dennissiemensma, 2020-04-01:
Ter vergelijking, ik heb ook een v4 meter, dit is voor mij het verloop elke minuut:

Het loopt elke minuut op totdat er minstens 1 meting in de volgende minuut is en de systeemtijd in DSMR-reader is de minuut gepaseerd. Dan gaat die groeperen/verwerken.

```
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading where processed = false order by timestamp desc;
    id    |       timestamp        
----------+------------------------
 13274099 | 2020-04-01 18:14:53+01
 13274098 | 2020-04-01 18:14:43+01
 13274097 | 2020-04-01 18:14:32+01
 13274096 | 2020-04-01 18:14:22+01
 13274095 | 2020-04-01 18:14:12+01
 13274094 | 2020-04-01 18:14:02+01
(6 rows)
```

```
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading where processed = false order by timestamp desc;
    id    |       timestamp        
----------+------------------------
 13274100 | 2020-04-01 18:15:03+01
 13274099 | 2020-04-01 18:14:53+01
 13274098 | 2020-04-01 18:14:43+01
 13274097 | 2020-04-01 18:14:32+01
 13274096 | 2020-04-01 18:14:22+01
 13274095 | 2020-04-01 18:14:12+01
 13274094 | 2020-04-01 18:14:02+01
(7 rows)
```

```
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading where processed = false order by timestamp desc;
    id    |       timestamp        
----------+------------------------
 13274100 | 2020-04-01 18:15:03+01
(1 row)
```

jghaanstra, 2020-04-01:
> Bedankt voor de uitgebreide toelichting. Er zijn diverse dingen die hier in kunnen meespelen, maar ik zie zelf niet zo 1-2-3 een verband tussen de inhoud van de gegevens en het wel of niet verwerken.
> Het enige wat daar invloed op heeft zijn de systeemtijden van de slimme meter en die van het OS waar DSMR-reader (of de database) op draait.

Ah, dit zou het wel eens kunnen zijn. Home Assistant doet een POST request op het moment dat er een nieuw DSMR bericht binnenkomt (dit is inderdaad elke 10 seconden). In de data gebruik ik echter als timestamp de timestamp waarop de consumptie in Home Assistant is bijgewerkt. Maar dit loopt natuurlijk spaak op het moment dat er alleen productie is. Ik ga dit aanpassen en zal daarvan morgen de resultaten zien.

Maar ik vermoed dat ik daarmee dit issue kan sluiten. Bedankt Dennis en Bram.

dennissiemensma, 2020-04-01:
Bedankt voor de opheldering. Mocht je er toch niet uit komen dan hoor ik het later alsnog graag.

jghaanstra, 2020-04-02:
Helaas, ook met de aanpassing van de juiste timestamp blijft dit probleem zich voordoen. Ik zal binnenkort de genoemde queries op de database draaien om te zien of dat meer informatie oplevert.

jghaanstra, 2020-04-02:
Zoals het nu lijkt wordt de timestamp van API calls waarbij `electricity_currently_delivered` = 0.000 genegeerd en wordt in de database bij deze readings de timestamp vastgelegd van de laatste call waarbij `electricity_currently_delivered` > 0 is geregistreerd. Aangezien dit leidt tot meerdere records met dezelfde timestamp worden deze niet verwerkt. Dit duurt voort totdat er een API call wordt gedaan met `electricity_currently_delivered` > 0 en er dus weer records worden toegevoegd met een nieuwe timestamp. Hieronder het resultaat van de tabel. Van 15:17 tot 15:22 was `electricity_currently_delivered` = 0.000 en om 15:22 was `electricity_currently_delivered` > 0 waarna de timestamps weer werden bijgewerkt. Lijkt dus toch een bug te zijn in DSMR Reader.

```
118566	"2020-04-02 15:23:28.582312+00"	false
118565	"2020-04-02 15:23:18.557962+00"	false
118564	"2020-04-02 15:22:58.484671+00"	true
118563	"2020-04-02 15:22:48.480518+00"	true
118562	"2020-04-02 15:22:38.452952+00"	true
118561	"2020-04-02 15:22:38.452952+00"	true
118560	"2020-04-02 15:17:17.678619+00"	true
118559	"2020-04-02 15:17:17.678619+00"	true
118558	"2020-04-02 15:17:17.678619+00"	true
118557	"2020-04-02 15:17:17.678619+00"	true
118556	"2020-04-02 15:17:17.678619+00"	true
118555	"2020-04-02 15:17:17.678619+00"	true
118554	"2020-04-02 15:17:17.678619+00"	true
118553	"2020-04-02 15:17:17.678619+00"	true
118552	"2020-04-02 15:17:17.678619+00"	true
118551	"2020-04-02 15:17:17.678619+00"	true
118550	"2020-04-02 15:17:17.678619+00"	true
118549	"2020-04-02 15:17:17.678619+00"	true
118548	"2020-04-02 15:17:17.678619+00"	true
118547	"2020-04-02 15:17:17.678619+00"	true
```

dennissiemensma, 2020-04-02:
Bedankt voor je update. Zoals eerder aangegeven is het tijdstip bepalend voor de verwerking. DSMR-reader doet niets totdat de volgende meting binnen is.

Tevens geef je hierboven aan:
>  In de data gebruik ik echter als timestamp de timestamp waarop de consumptie in Home Assistant is bijgewerkt

Dat moet eigenlijk het tijdstip zijn wat in het telegram staat. Daarnaast kun je zien aan de response van de API welk ID en timestamp er is weggeschreven.

Ik heb het hier nog een keer extra getest, er vanuitgaande dat je de v2 API `/api/v2/datalogger/dsmrreading` gebruikt. Als timestamp parameter geef ik `2020-01-15T12:34:56+01` en de response die ik krijg is:
```
{
    "id": 13269366,
    "timestamp": "2020-01-15T12:34:56+01:00",
    "electricity_delivered_1": "2000.000",
    "electricity_returned_1": "0.000",
    "electricity_delivered_2": "3000.000",
    "electricity_returned_2": "0.000",
    "electricity_currently_delivered": "1.500",
    "electricity_currently_returned": "0.025",
    "phase_currently_delivered_l1": null,
    "phase_currently_delivered_l2": null,
    "phase_currently_delivered_l3": null,
    "extra_device_timestamp": null,
    "extra_device_delivered": null,
    "phase_currently_returned_l1": null,
    "phase_currently_returned_l2": null,
    "phase_currently_returned_l3": null,
    "phase_voltage_l1": null,
    "phase_voltage_l2": null,
    "phase_voltage_l3": null,
    "phase_power_current_l1": null,
    "phase_power_current_l2": null,
    "phase_power_current_l3": null
}
```

Als ik vervolgens in de database kijk, is het tijdstip exact hetzelfde (welliswaar in UTC): `2020-01-15 11:34:56.000000 +00:00`. Dus als je hetzelfde tijdstip ziet over een langere periode, dan is het heel aannemelijk dat dat tijdstip verkeerd wordt aangeleverd.

In je database is te zien dat er een gat van 5 minuten zit tussen 118560 en 118561. Normaal gesproken betekent dat dat DSMR-reader 5 minuten geen telegram heeft ontvangen. **Echter, gezien je zelf het tijdstip bepaalt via de API, kan ik niet zeggen of er daadwerkelijk 5 minuten niets is binnengekomen in DSMR-reader of dat het tijdstip 5 minuten niet verandert.** Hoe dan ook zit er daar iets scheef, want hiermee krijg je eigenlijk dezelfde situatie als wanneer je 5 minuten lang de seriele poort niet uitleest.

Ik raad je aan om gedurende die 5 minuten regelmatig de query uit te voeren om te kijken welke tijdstippen er toegevoegd worden (en of er dus nieuwe metingen bij komen). Als het tijdstip wederom **hetzelfde blijft**, dan betekent het dat HA het timestamp in de API verkeerd meegeeft of niet bijwerkt. Zie ook mijn eerdere dump van hoe het verloop normaal is met een meter.

Hopelijk geeft je dit meer aanwijzingen om uit te zoeken waar het exact mis gaat.

jghaanstra, 2020-04-03:
Argh, shame on me. Ik had de aanpassingen voor de timestamp wel doorgevoerd maar Home Assistant niet herstart. Hierdoor werd nog steeds de timestamp van consumptie meegeleverd bij de productie API calls. Door de juiste timestamps mee te leveren is het probleem opgelost. Sorry voor de moeite.

closed 2020-04-03 by jghaanstra