dsmrreader/dsmr-reader #949: ISKA AM550 geen telegrammen

remyderuysscher, 2020-04-01:
dsmr-reader 3.7 (docker)
Synology NAS
SMR v5 / ISKA AM550

Heb een geschikte kabel gekocht voor de meter, krijg echter geen telegrammen binnen.
Zie geen foutmeldingen, kan het te maken hebben met de instellingen van de seriele poort?

dennissiemensma, 2020-04-01:
Bedankt voor je melding. Er is een soortgelijk issue met Docker Synology in #929. Alleen in die situatie stopte de seriele poort na een tijd.

Voor inhoudelijke vragen voor Docker kun je dit het beste bij xirixiz zelf leggen, er vanuitgaande dat je de Docker-image van xirixiz gebruikt. Zie ook de readme daar, wellicht is dit al voldoende: `sudo usermod -aG dialout $(whoami)`.

remyderuysscher, 2020-04-01:
Ik gebruik de ualex73 container.

Ok want ik heb ook een andere docker container op Synology met een andere P1 kabel op een Landis+Gyr DSMR 5.0 meter die werkt wel.

Welke seriele instelling moet ik voor de AM550 toepassen?

dennissiemensma, 2020-04-01:
Ik heb helaas geen kant en klare lijst voor je welke meter welke instellingen gebruikt. Je kunt het beste even kijken of je `cu` kunt gebruiken om uit te vogelen welke combinatie werkt, al weet ik niet of dat ook makkelijk te doen is op een NAS: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional

Verder zou je nog even kunnen zoeken op 'AM550' in dit topic: https://gathering.tweakers.net/forum/list_messages/1578510
Of daar eventueel om advies vragen. Mochten de standaard-opties in DSMR-reader niet de juiste seriele-instellingen ondersteunen, kan ik die altijd nog voor je uitbreiden.

remyderuysscher, 2020-04-03:
Follow up, het werkt. 

Pin1 Rood -> 5V
Pin2 Groen -> RTS
Pin3 Zwart -> GND
Pin5 Wit -> RXD

Pull-up weerstand 10K tussen RTS en RXD.  De baudrate stond goed in de ualex73 docker container.

closed 2020-04-03 by remyderuysscher