dsmrreader/dsmr-reader #955: Resetten van meter statistieken

n9iels, 2020-04-02:
Ik denk dat het kunnen resetten van de meter statistieken en de _All time highs & lows_ een nuttige feature zou kunnen zijn. Momenteel staat mijn all time low allemaal op 1 Watt en heb ik er dus vrij weinig aan. Daarnaast kan het nuttig zijn als je eens een keer een soort nul-meting wilt doen om te kijken hoeveel energie je in e nacht verbruikt of iets dergelijks.

dennissiemensma, 2020-04-02:
Bedankt voor je verzoek. Het resetten van de meterstatistieken zou geen zin hebben, want die komen met elk telegram opnieuw binnen.

Wat betreft de highs en lows. Deze kun je nu ook al zien in `/admin/dsmr_stats/electricitystatistics/`, alleen zijn ze nog read-only. Ik zal ze bewerkbaar maken, zodat je ze zelf eventueel kan wijzigen voor een reset of nachtmeting.

Mocht je zonnepanelen hebben, dan zeggen die all-time lows niet zoveel omdat die dan vervuild raken door de opname en de slimme meter kan dat niet detecteren.

dennissiemensma, 2020-04-02:
Toegevoegd aan volgende release.

closed 2020-04-02 by dennissiemensma