dsmrreader/dsmr-reader #956: Fout Django

D1StrX, 2020-04-02:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: 
v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi ...
RaspberryPi 3b
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*:
Native

*Issue*:
Heb met reboots geprobeerd DSMR te redden, wat werkten de eerste keer. Supervisorctl zegt dat alles running is, maar wederom krijg ik de error onderstaand. 
Zag dat dit al een keer eerder [gepost ](https://github.com/dennissiemensma/dsmr-reader/issues/476) is, alleen werd toen gezegd dat de PostgreSQLdb niet draait. En verder geen commentaar. 

Pi geupdate met nieuwste updates/firmware voor Ubuntu 18.04.4 for Pi & vervolgens DSMR geupdate naar v3.7. De onderstaande error werd al weergeven op v3.6, vandaar ook geupdate naar v3.7 in de hoop het weer werkende te krijgen. 

Error: 

Server Error
Sorry, something unexpected happened.
Exception:
OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 18, in get_context_data
    context_data['capabilities'] = dsmr_backend.services.backend.get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 36, in get_capabilities
    'electricity': ElectricityConsumption.objects.exists(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 777, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/query.py", line 537, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1121, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1149, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 260, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 236, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 197, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

https://github.com/dennissiemensma/dsmr-reader/issues/476

dennissiemensma, 2020-04-02:
Bedankt voor je melding. De database lijkt niet bereikbaar en dat lijkt deels op #925. 

Je kunt kijken of de database draait met:

```
sudo systemctl status postgresql
```

En je kunt kijken of je handmatig in de database kan:

```
sudo su - postgres
psql dsmreader
```

Laat maar weten wat de output is.

D1StrX, 2020-04-03:
Heb dit helaas niet meer kunnen doen. Na een paar pogingen tot herstel (waaronder db proberen terug te zetten via backup methode)/reboots kon de back-end en de andere services die supervisorctl weergeeft de pid files niet meer vinden/waren steeds aan het herstarten. 

En daarbij was me opgevallen dat de sd-kaart van 32gb helemaal volgelopen was. 

Heb een backup kunnen overzetten, een verse installatie op de Pi uitgevoerd en vervolgens backup restored. Nu draait Dsmr weer.

Toch bedankt voor de hulp en snelle reactie.

closed 2020-04-03 by D1StrX

dennissiemensma, 2020-04-03:
Fijn om te horen. 

Je kunt overigens retentie instellen, zodat brongegevens na een tijd opgeschoond worden: https://dsmr-reader.readthedocs.io/en/latest/admin/datalogger.html#data-cleanup-retention
Vooral bij DSMR-v5 meters gaat het hard.

Een ander ding wat je kunt doen, is de sleep van de datalogger verhogen in de admin-interface. Dan krijg je minder telegrammen binnen en groeit je data ook minder hard, terwijl het voor de dagtotalen niets uitmaakt.