dsmrreader/dsmr-reader #958: DMSR data logger 'loopt vast' na een minuut of 2 na upgrade van 3.5 naar 3.8

midget1234, 2020-04-03:
Na mijn upgrade naar 3.8 ismijn dsmr set-up zeer instabiel. De data logger stopt ermee kort na de start. Ik heb geen idee waar het aan kan liggen. Ik heb geen veranderingen in hardware oid doorgevoerd. Hulp is welkom.

Running on Pi3 
Landis & Gyr DMSR 5

dennissiemensma, 2020-04-04:
Bedankt voor je melding. 

- Draai je het native of in Docker?
- Wat zie je in de logfiles? https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html#logging

midget1234, 2020-04-04:
Hallo Dennis, 

Super bedankt voor je snelle reactie. 

Ik draai DSMR native op een Raspberry Pi.

Ik heb de logs zojuist aangezet conform de link die je gaf.

Het lijkt erop dat er iets mis gaat met de datalogger wat die kan niet herstart worden (zie screen print). 
Ik zal het systeem weer rebooten zodat ie het weer even doet en dan kan ik kijken wat er in de logfiles staat,

Super bedankt voor je support!

Groeten uit Arnhem, Ruud Somers

midget1234, 2020-04-04:
ruud@wytske.nl sent you some files

1 item, 4.43 MB in total ・ Will be deleted on 11 April, 2020

Download link:
https://wetransfer.com/downloads/864e4cfcd1799654ba4bab045f54206320200404201141/f799109f8166a1883d40ba1f8c4fca0420200404201200/c70553

Message:
dsmrreader.log

1 item

dsmrreader.log - 4.43 MB

[Get more out of WeTransfer, get Pro](https://wetransfer.com/pro?trk=WT201704_email&utm_campaign=WT_email_tracking&utm_content=general&utm_medium=plus_footer_ad_link&utm_source=notify_recipient_email)

About WeTransfer: https://wetransfer.com/about
Help: https://wetransfer.zendesk.com/hc/en-us
Legal: https://wetransfer.com/legal/terms

To make sure our emails arrive, please add noreply@wetransfer.com to [your contacts](https://wetransfer.zendesk.com/hc/en-us/articles/204909429).

midget1234, 2020-04-04:
Hallo Dennis bij deze mijn log bestanden; 

De dsmrreader.log heb ik gestuurd met wedtransfer (4.7mb)

Hopelijk kun je hier aan zien wat er verkeerd gaat. Ik kan er eigenlijk geen touw aan vast knopen wat er in de log files staat.

De applicatie draaide als een zonnetje op v3.5 en zonder overige wijzigingen stopt de applicatie na de upgrade naar v3.8

Wellicht nog goed om te vermelden dat ik op de pi ook nog ser2net draai om zo de DSMR gegevens van de pi over te zetten naar Home Assistant (die draait in docker op my Synology NAS). 

Echt een fijn programma DSMR om inzicht te krijgen in verbruik en energie productie.

Met vriendelijke groet, 

Ruud Somers

dennissiemensma, 2020-04-04:
In deze log staan wat geweigerde telegrammen, maar dat is normaal. Ik ben vooral benieuwd naar de dsmr_datalogger-log zelf, gezien je aangeeft dat dat proces stopt.

midget1234, 2020-04-06:
Hallo Dennis,

De volgende vreemde situatie doet zich voor;

Op de website van de DSMR zie ik regelmatig dat het laatste telegram lang
geleden is verstuurd (bijvoorbeeld nu 6 uur sinds laatste telegram, zie
Status screen print). Echter via mijn Home Assistant Dashboard zie ik wel
steeds de actuele readings per seconde veranderen. Zonder tussenkomt of
resetten of iets dergelijks gaat de DSMR website soms weer werken (zie ik
iedere paar seconden een telegram en is de status OK) om vervolgens weer
urenlang 'vast te lopen'. Erg vreemd.

De datalogger werkt dus gewoon en stuurt via ser2net de gegevens door naar
Home Assistant op de NAS (zie HA screenprint) maar de DSMR website kan niet
weergeven wat de datalogger daadwerkelijk doet.

Welke logfile(s) heb je nodig om dit te kunnen bekijken ?

Met vriendelijke groet, Ruud Somers

[image: image.png]

[image: image.png]

dennissiemensma, 2020-04-06:
Ik denk dat je dit het beste kunt zoeken in `/var/log/supervisor/dsmr_datalogger.log`.

Hoe komt de data binnen bij DSMR-reader? Via HA? Ik heb zelf geen ervaring met ser2net.

midget1234, 2020-04-08:
Hallo Dennis, 

In deze logfile staat niet iets dat wijst op een foutmelding…

dennissiemensma, 2020-04-09:
Hoe komt de data binnen bij DSMR-reader? Via HA? Ik heb zelf geen ervaring met ser2net.

midget1234, 2020-04-09:
Hallo Dennis de data komt rechtstreeks bij DSMR binnen. 

De P1 van mijn Landis & Gyr meter zit aangesloten op een Raspberry Pi. Op deze Pi draaien 2 programma’s; DSMR v3.8 en Ser2net. Ser2Net stuurt de telegrammen door naar HA (in een docker op een Synology NAS).

Het doorsturen via Ser2net gaat goed en ik zie het ‘live’ verbruik/productie in HA.

Wat niet goed gaat is dat DSMR )op de Pi) aangeeft dat het laatste telegram 4 uur geleden is ingelezen, of 10 minuten geleden.. steeds een storing die ook vanzelf weer verholpen is en dan weer opnieuw optreed.

Heel vreemd..

Groeten, Ruud

dennissiemensma, 2020-04-10:
Begrijp ik het goed dat zowel DSMR-reader als Ser2net dezelfde seriele poort uitlezen? Of kloon je die nog met socat of iets?

midget1234, 2020-04-11:
Volgens mij wel. Nee, ik kloon niets.

In v3.5 ging dit zonder problemen. In v3.8 niet.

Gr, Ruud

dennissiemensma, 2020-04-11:
Voor zover ik weet is het niet mogelijk dat twee applicaties dezelfde seriele poort uitlezen. Wat je dan gaat krijgen is dat de ene de andere er telkens uitgooit (of er niet inkomt).
Hier is namelijk al eerder een issue voor gemaakt en daar staat ook een workaround in, door het met socat te klonen: Zie #377.

Dat zeggende hebben, kun je het beste controleren of dit hetzelfde probleem is door simpelweg ser2net even uit te zetten, de DSMR-datalogger te herstarten en dan te kijken of die nog steeds wegvalt.

dennissiemensma, 2020-04-28:
@midget1234 is dit inmiddels opgelost?

midget1234, 2020-04-29:
Hallo Dennis, nee niet opgelost, wel werkbaar: DSMR informatie 'hapert' (is
slechts bij 'toeval' beschikbaar) maar de datavoorziening naar Home
Assistant werkt wel goed. Doordat HA mijn belangrijkste monitoring bron is
stoort het me niet.

Bedankt, Ruud Somers

dennissiemensma, 2020-04-30:
Oke prima. Mocht je er later nog mee aan de slag willen, dan raad ik je aan om de poort te klonen met socat. Ik verwacht dat dat het oplost in dit geval.

closed 2020-04-30 by dennissiemensma

midget1234, 2020-04-30:
Merci, ga ik doen!

Verstuurd vanaf mijn iPhone