dsmrreader/dsmr-reader #959: de upload naar pvoutput stuurt geen generated data

kampie8, 2020-04-06:
Hoi Dennis,

Afgelopen weekend heb ik geupdate naar de laatste versie van dsmr. Nu dacht ik dat het leuk zou zijn om pvoutput te vullen met data vanuit dsmr, maar helaas gaat daar iets fout. vorig jaar toen ik net panelen had heb ik het  1 keer werkend gehad en dat data naar pvoutput ging.
in de log van dsmr staat dit:

[2020-04-06 11:45:11,943] WARNING   [i] PVOutput: Data found, not in sync. Last data timestamp < expected (2020-04-06 11:44:00+02:00 < 2020-04-06 11:45:00+02:00

op pvoutput ben ik te vinden op KAMPIESOLAR.

heb je enig idee wat te doen? 

groeten Raymond

dennissiemensma, 2020-04-06:
Bedankt voor je melding. Zou je debug-logging willen aanzetten om te kijken wat er nog meer gebeurt?

Zie https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html#logging

kampie8, 2020-04-06:
Hoi Dennis,

Hierbij de logging met debug aan. kun je hier meer mee?

[dsmr_backend.log](https://github.com/dennissiemensma/dsmr-reader/files/4440291/dsmr_backend.log)

dennissiemensma, 2020-04-06:
Ik zie een hoop foutmeldingen met
```
column dsmr_backup_backupsettings.latest_backup does not exist
LINE 1: ...kup_time", "dsmr_backup_backupsettings"."folder", "dsmr_back...
```

Had je die zelf ook al gezien? Of zijn die van eerder gezien ze later niet meer voorkomen? 

Je kunt kijken of dit iets oplost, maar dat weet ik dus niet zeker:

```
sudo su - dsmr
./manage.py dsmr_sqlsequencereset
./manage.py migrate
```

dennissiemensma, 2020-04-06:
Verder zie ik wel af en toe dit in de logs:

```
[2020-04-06 20:40:14,636] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-04-06 20:40:29.623307+02:00 (ETA 0:00:14.987062)
[2020-04-06 20:40:14,741] DEBUG    PVOutput: Uploading data: {'d': '20200406', 't': '20:40', 'v3': 6559, 'v4': 750, 'n': 1}
[2020-04-06 20:40:15,378] DEBUG    PVOutput:  Delaying the next export until: 2020-04-06 20:45:00+02:00
```

Kan het zijn dat jij bedoelt met 'generated data' je teruglevering? Want dat geeft DSMR-reader inderdaad bewust niet door aan PV-output.

dennissiemensma, 2020-04-06:
Ik kan je trouwens aanraden om in admin > backend configuratie > de sleep voor de backend wat hoger te zetten, bijvoorbeeld op 5 of 10 seconden. Dat spamt je systeem wat minder.

kampie8, 2020-04-07:
Hoi Dennis, 

De sleep nu idd op 5 sec gezet. Die backup fout is gekomen doordat ik een oude topic van je vond, waar je met check branch aan de gang ging naar een regel met 456 of iets erin, toen heb ik hem terug gezet naar de master omgeving, alleen zat ik daarna weer op 2.15. Daarna de hele procedure naar v3 weer uitgevoerd en daarbij ging de backup fout, vandaar de melding.
Dit dat wat jij ziet, dat klopt, maar dit genereert alleen maar een verbruiksgrafiek op binnen pvoutput. ik wil ook de groene zien voor teruglevering, of ondersteund dsmr_reader dit niet, want je zegt dat generated data niet wordt verzonden. Ik heb het idee dat er nu te weinig in de upload zit, want ik stuur ook de data naar een MQTT intern hier, en daar krijg ik alle meterstanden door, ook de teruglevering ivm de zonnepanelen. 
Ondertussen komt de temperatuur ook door op pvoutput.org, maar dat komt door mijn weerstation. ik stuur een screenshot mee.
[image]

dennissiemensma, 2020-04-07:
Klopt. In dat geval was er een misverstand waar we het over hadden. DSMR-reader geeft de teruglevering niet door, omdat die niet klopt, want een deel is al opgenomen in je huishouden. Het is dan netto, net zoals het verbruik, en eigenlijk zie je dan in PVOutput exact hetzelfde als in DSMR-reader en dat voegt weinig toe.

Zoals het ooit bedoeld was, kun je het beste topic #9 doorlezen. Als ik het me goed herinner is het de bedoeling dat je omvormer direct de bruto oplevering naar PVOutput stuurt. DSMR-reader stuurt dan het netto-verbruik door naar PVOutput en dan is in PVOutput daar wat mee te doen. Dat was ook de reden dat de integratie destijds is gemaakt.

Voor algemene gegevensexport van DSMR-reader kun je verder de API (pull) of MQTT (push) gebruiken. Of een plugin maken die iets anders naar PVOutput stuurt: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-1-upload-data-to-second-pvoutput-account

closed 2020-04-07 by dennissiemensma