dsmrreader/dsmr-reader #961: Nog geen 3.8 versie beschikbaar voor Docker?

niekniek89, 2020-04-08:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Intel NUC
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Docker
* DSMR-protocol slimme meter *(v4 of v5)*: v4

Goedemorgen,

Ik heb zojuist binnen Docker "recreate" aangeklikt, en deze word netjes geupdate.
echter word alleen nog versie 3.7 aangeboden als latest, en niet de echte laatste versie (v3.8).
kan het kloppen dat v3.8 nog niet klaar is voor docker? of moet ik hiervoor bij iemand anders zijn?

alvast bedankt.

dennissiemensma, 2020-04-08:
Daarvoor zul je bij de aanbieder van die DOcker container moeten zijn. Ik zie de v3.8 tags wel in de hubs:

- https://hub.docker.com/r/ualex73/dsmr-reader-docker/tags
- https://hub.docker.com/r/xirixiz/dsmr-reader-docker/tags/

closed 2020-04-08 by dennissiemensma

niekniek89, 2020-04-09:
Bedankt voor je reactie.

ik gebruik de tag van https://hub.docker.com/r/xirixiz/dsmr-reader-docker/tags/.

echter werkt de "tag" latest niet. Zie onderstaande opmerking van Xirixiz:

I notice there is an issue indeed with this tag... I'll check the Azure DevOps pipeline what went wrong and get back to you.