dsmrreader/dsmr-reader #962: [#937] Dashboard €/uur houdt geen rekening met teruglevering

niekniek89, 2020-04-09:
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.8
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Intel NUC
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Docker
* DSMR-protocol slimme meter *(v4 of v5)*: v4

Goedemorgen,

ik heb zojuist de laatste versie van DSMR reader geinstalleerd.
Bij het terugleveren zie ik nu dat deze niet meer op 0.00 €/uur staat, maar oploopt, terwijl ik terug lever (zie screenshots). 

Terwijl ik terug lever, loopt het totaal bedrag wel op, terwijl deze eigenlijk stil moet blijven staan tijdens het terug leveren?

gaat er wellicht bij mijn DSMR reader iets niet goed?

[image]
[image]

niekniek89, 2020-04-09:
Zat een fout in mijn DSMR settings (qua tarieven leverancier).
Versie 3.8 draait prima, niks mis mee.
Excuus, was een foutje van mijzelf.

closed 2020-04-09 by niekniek89

dennissiemensma, 2020-04-09:
Bedankt voor je terugkoppeling. Fijn dat je het hebt kunnen oplossen.