dsmrreader/dsmr-reader #963: Kosten Dashboard versie 3.8

niekniek89, 2020-04-10:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.8
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Intel NUC
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Docker
* DSMR-protocol slimme meter *(v4 of v5)*: v4

Goedemiddag,

ik heb toch nog een vraag over het dasboard gedeelte.
sinds versie 3.8 zie ik tijdens het leveren van energie (zonnepanelen) de bedragen netjes teruglopen.
echter lijkt het alsof het totaalbedrag daarbij niet meer klopt (zie screenshot).

het lijkt alsof hij het gas gedeelte ook herberekend, tijdens het terugleveren van energie.
kan dit kloppen?

je zou zeggen dat het bedrag voor het gas gedeelte zo zou moeten blijven.

[image]

dennissiemensma, 2020-04-10:
Ik begrijp niet helemaal wat je bedoelt. Als ik de bedragen nareken, lijkt het totaal te kloppen:
```
0,31 − 0,49 + 0,17 = -0,01
```

Daarnaast heb ik even de variabelen voor gas gecheckt in de template, dat lijkt ook goed te staan.

Kun je meer toelichting geven?

niekniek89, 2020-04-11:
Goedemiddag @dennissiemensma,

Bedankt voor je reactie.
Je hebt helemaal gelijk, ik zit verkeerd te kijken (dom)!
Heb zelf een rekenfoutje gemaakt, maar begrijp nu hoe DSMR omgaat met de berekening.

Fijne dagen nog!

closed 2020-04-11 by niekniek89