dsmrreader/dsmr-reader #964: Onverwerkte telegrammen

lukevredeveld, 2020-04-11:
Mijn DSMR-reader is vanmorgen rond een uurtje of 6 gestopt met het verwerken van Telegrammen. Omdat er een nieuwe versie klaar stond, heb ik eerst een update gedaan. Daarna liep het aantal telegrammen nog steeds op. Daarna heb ik alle processen nog een aantal malen herstart met `sudo supervisorctl restart all`

Als de processen opnieuw gestart zijn, dan werkt DSMR-reader gedurende enkele minuten een paar honderd van de 28.000 achterstallige telegrammen weg. Na een paar minuten loopt het aantal onverwerkte telegrammen echter weer op. Dit is de weergave van het log-file van het backend proces.

[image]

dennissiemensma, 2020-04-11:
Bedankt voor je melding. Die foutmelding van PVoutput is vrij onschuldig en lost zichzelf op. Ik zou graag deze gegevens nog wat je willen weten:

**Wat gebruik je?**
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi ...
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: ...
* DSMR-protocol slimme meter *(v4 of v5)*: v...

dennissiemensma, 2020-04-11:
Verder kun je waarschijnlijk debug-logging aanzetten om te zien wat het systeem (de backend) precies doet: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging

dennissiemensma, 2020-04-11:
Daarnaast kun je nog even topic #913 doorlezen voor aanwijzigingen. Zodra je meer informatie hebt kunnen we wat gerichter zoeken.

lukevredeveld, 2020-04-11:
Oh sorry, was inderdaad een beetje te snel met mijn post. Had eerst alles gewist en was toen vergeten die gegevens die je vraagt toe te voegen :-)

Het is een pi 3b met native installatie. Ik kan ff niet vinden welke dsmr-protocol slimme meter ik gebruik.... waar kan ik dat zien?

Logging ga ik zometeen aanzetten als de pi klaar is met alle andere pakketten updaten en upgraden.

dennissiemensma, 2020-04-11:
> Het is een pi 3b met native installatie. Ik kan ff niet vinden welke dsmr-protocol slimme meter ik gebruik.... waar kan ik dat zien?

Als je elke seconde een telegram krijgt heb je DSMR v5, als je elke 10 seconden een telegram krijgt heb je DSMR v4.

Dat kan namelijk van invloed zijn op de performance. Als je v5 hebt is mijn advies om sowieso de datalogger wat te laten vertragen, zodat je 5 of 10x minder telegrammen binnenkrijgt. Dat scheelt je performance en opslag, terwijl de dagtotalen er niet door beinvloed worden.

lukevredeveld, 2020-04-11:
Dan heb ik v5 :-)

Overigens is deze PI dedicated voor DSMR met een 32Gb kaartje erin. Dus het kan wel wat hebben. Ik heb ook gecheckt en de kaart zit nog lang niet aan zijn max.

dennissiemensma, 2020-04-11:
Het gaat niet alleen om de opslag, maar ook of de hardware het kan bijbenen qua verwerken.

Hoe dan ook kun je het beste de backend-log in debugging mode bekijken. Dan zie je precies wat die doet, waar die op wacht en wat er (relatief) lang duurt.

lukevredeveld, 2020-04-11:
Yep. Ga ik zo aanzetten. Ik had blijkbaar een sh*tload aan achterstallige updates op die PI uit te voeren, want hij heeft net 788MB binnengehaald. Is nu op 3% update...

lukevredeveld, 2020-04-11:
Oh... dit is niet goed. Ik heb de debug aangezet. Daarna alles dus opnieuw gestart, conform de handleiding. Toen ging DSMR-reader weer telegramme wegwerken. Maar nu dus dit:

`
Server Error
Sorry, something unexpected happened.
Exception:

OperationalError: FATAL:  the database system is shutting down
FATAL:  the database system is shutting down

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/status.py", line 12, in get_context_data
    context_data['status'] = dsmr_backend.services.backend.status_info()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 193, in status_info
    'electricity': get_electricity_status(capabilities),

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 101, in get_electricity_status
    status['latest'] = ElectricityConsumption.objects.all().order_by('-pk')[0].read_at

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 313, in __getitem__
    qs._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1261, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 57, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1149, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 260, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 236, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 197, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

`

dennissiemensma, 2020-04-11:
Het lijkt erop dat er iets met de database is. Je kunt het systeem een keer proberen te herstarten en anders even in de logs van postgresql kijken.

lukevredeveld, 2020-04-12:
Na de update van de PI heb ik DSMR-reader een aantal keren herstart, waarna hij weer goed begon te lopen. Na een half uurtje waren alle telegrammen weer verwerkt. De rest van de dag / avond is hij goed blijven lopen. Vannacht is hij echter weer opeens gestopt met het verwerken van telegrammen. De LOG-file van de backend stopt ook gewoon op dat moment (zie hieronder). Kun je me aangeven hoe ik de datalogger kan vertragen? Misschien dat dat nog iets oplevert.

[image]

dennissiemensma, 2020-04-12:
Als in je de admin-interface naar Backend configuratie gaat kun je daar de sleep instellen voor de backend. Die kun je op 5 seconden zetten.

Voor de datalogger sleep kun je in de admin naar Datalogger configuratie en daar de sleep instellen voor de datalogger. Je kunt daar ook eerst 5 seconden of meer proberen.

----

Alleen zou het systeem niet zomaar moeten stoppen, hooguit wat trager moeten zijn als die het niet aankan. Staan er in logfiles van `/var/log/supervisor/` verder fouten? Supervisor zou sowieso oneindig moet proberen de processen weer te starten.

lukevredeveld, 2020-04-13:
De backend logg op de locatie /var/log/supervisor laat ook gewoon een 'gat' zien op het moment dat het probleem zich voordoet (zie hieronder). 

Overigens heb ik de sleep voor de datalogger nu op 5 gezet (stond op 0.5). Sindsdien blijft het aantal onverwerkte telegrammen <10 staan. Dus so far, so good... wel vreemd van die logfiles?

[image]

dennissiemensma, 2020-04-14:
Bedankt voor de terugkoppeling. Als het verder nu wel goed werkt zou ik het even aankijken of dat een paar dagen goed blijft gaan.

lukevredeveld, 2020-04-18:
Het bleef inderdaad 4 dagen goed gaan... inmiddels wordt er weer geen telegram verwerkt en sta ik er alweer 22.000 achter.... nog ideeen?

dennissiemensma, 2020-04-19:
Wat staat er in de logs? Loopt de achterstand wel in of wordt die alleen maar groter?

dennissiemensma, 2020-04-28:
@lukevredeveld ben je hier nog verder mee gekomen?

lukevredeveld, 2020-04-28:
Als hij stopt met verwerken dan start ik de boel opnieuw op en dan gaat hij weer inlopen. Hij gaat inmiddels al weer 10 dagen goed. Hopen dat het zo blijft

dennissiemensma, 2020-04-28:
Prima. Ik ga er vanuit dat het verhogen van de sleep-tijden het wellicht heeft opgelost. En anders kun je er hier later nog op terugkomen.

closed 2020-04-28 by dennissiemensma