dsmrreader/dsmr-reader #965: Error: could not connect to server: Connection refused

Kludon, 2020-04-12:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.8
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v4

Na de installatie van 3.8 werkte de applicatie. Daarna heb ik mijn pi geupdate en kreeg ik DSMR niet meer aan de praat met onderstaande meldingen. Hoe kan ik DSMR weer opstarten ? 

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

Server Error
Sorry, something unexpected happened.
Exception:
OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 18, in get_context_data
    context_data['capabilities'] = dsmr_backend.services.backend.get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 36, in get_capabilities
    'electricity': ElectricityConsumption.objects.exists(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 777, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 537, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1121, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1149, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 260, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 236, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 197, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

Kludon, 2020-04-12:
I had also changed my language settings to nl-nl. Changed it back to en-us and the app is running.
But at some point I now get 504 Gateway Time-out if I want to read my status or my dsmr readings

dennissiemensma, 2020-04-12:
Bedankt voor je melding. Het lijkt erop dat DSMR-reader niet meer bij de database kan.

- Wat is de output van deze twee commands?

```
lsb_release -a

sudo su - postgres
psql dsmrreader
```

- Staan er foutmeldingen in de log van PostgreSQL?
```
# Mogelijk heet je file iets anders als je een andere versie draait.
sudo view /var/log/postgresql/postgresql-11-main.log
```

dennissiemensma, 2020-04-12:
Als de database wel goed werkt, kun je even in de logs kijken: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#supervisor

Staan daar fouten?

Kludon, 2020-04-12:
Hoi Dennis, door het wisselen van de language settings kon hij niet meer bij de database. door het terugzetten kan hij er weer bij. De gateway time out komt denk ik doordat ik gee nretentie heb teogepast. Ik heb er nu 2 jaar aan data instaan. Ik heb de retentie nu gezet op een jaar om een eerste deel te verwijderden maar nu loopt de applicatie weer vast omdat dat natuurlijk veel werk is. Is er een command line commando dat ik kan uitvoeren voor de rententie zodat  ik het kan monitoren ? Of is het nu gewoon wachten ?

dennissiemensma, 2020-04-12:
Bedankt voor je terugkoppeling. Dat is goed om te weten qua taalinstellingen, ik wist niet dat dat invloed had.

Wat betreft retentie, je kunt met debug-logging (https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging) kijken hoever die is en wat die doet. Dat kun je dan zien in de log van de backend.

Verder zou je nog kunnen kijken of de indexes op de tabellen staan: https://github.com/dennissiemensma/dsmr-reader/issues/913#issuecomment-593137228

dennissiemensma, 2020-04-12:
En als aanvulling op laatste comment, kun je deze tabel ook nog checken qua indexes:

```
\dS dsmr_consumption_electricityconsumption
```

Ter vergelijking
```
Indexes:
    "dsmr_consumption_electricityconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_electricityconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
    "dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" btree (phase_power_current_l1)
    "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" btree (phase_voltage_l1)
    "dsmr_consumption_electrici_phase_voltage_l2_917726c7" btree (phase_voltage_l2)
    "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" btree (phase_voltage_l3)
    "dsmr_consumption_electricityc_currently_delivered_5bf9be2d_uniq" btree (currently_delivered)
    "dsmr_consumption_electricityco_currently_returned_5f91fed7_uniq" btree (currently_returned)
    "dsmr_consumption_electricityconsumption_2241f8ee" btree (phase_currently_delivered_l2)
    "dsmr_consumption_electricityconsumption_5714c500" btree (phase_currently_delivered_l1)
    "dsmr_consumption_electricityconsumption_e90842a6" btree (phase_currently_delivered_l3)
```

dennissiemensma, 2020-04-28:
@Kludon ben je hier nog verder mee gekomen?

closed 2020-05-03 by dennissiemensma