dsmrreader/dsmr-reader #967: Het opnemen van watergebruik

allureGOT, 2020-04-12:
Is het een idee om het verbruik van water op te nemen?
Dit is volgens mij goed te doen met een pulsmeter.

dennissiemensma, 2020-04-12:
Bedankt voor je verzoek. Hoewel ik kan begrijpen dat dit heel handig is, beperkt dit project zich tot het uitlezen van het DSMR-protocol. 
Mocht het daarin opgenomen worden (de specificaties van Netbeheer NL houden er wel rekening mee), dan zal ik het zeker toevoegen.

closed 2020-04-28 by dennissiemensma

markminnoye, 2025-05-21:
Mijn Digitale meter geeft naast electriciteit en gas ook het waterverbruik door via DSMR. (België)
Ik merk op dat de HA plugin "DSMR Smart Meter" van netbeheer Nederland deze waarden _wel_ doorgeeft.
zou deze toegevoegd kunnen worden ?

[image]

Groeten,
Mark

dennissiemensma, 2025-05-21:
Zie #1813