dsmrreader/dsmr-reader #968: Database backup terugzetten

rickvnl, 2020-04-12:
Hi,

Helaas heb ik de SD kaart van mijn Pi gesloopt bij het verplaatsten van de Pi. 

Ik heb DSMR Reader al opnieuw geïnstalleerd op een andere SD kaart en de dropbox backup gedownload via Chromium. 

Ik weet nu niet goed hoe ik de backup terug moet zetten. 
De handleiding is hierin beknopt (https://dsmr-reader.readthedocs.io/nl/v3/installation/restore.html). Er staat dat er een lege database aangemaakt moet worden. 
Kan iemand mij vertellen hoe ik dit kan doen? 

Alvast bedankt!

dennissiemensma, 2020-04-13:
Als je DSMR-reader helemaal opnieuw hebt geinstalleerd:

```
sudo supervisorctl stop all

sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
```

Daarna kun je dit doen:
```
zcat pad/naar/backup.sql.gz | sudo sudo -u postgres psql dsmrreader
```

Waarbij je bovenstaand pad moet aanpassen naar waar je de backup op de Pi hebt gezet.

Als je verder geen fouten krijgt:
```
sudo supervisorctl start all

sudo su - dsmr
./deploy.sh
```

dennissiemensma, 2020-04-13:
Ik zet op de planning om de docs sowieso wat uit te breiden. Ze zijn hier idd vrij karig over zie ik.

rickvnl, 2020-04-13:
Super bedankt voor de instructies!
DSMR draait weer als een zonnetje!

closed 2020-04-13 by rickvnl