dsmrreader/dsmr-reader #969: In en uit knijpen van de grafieken werkt niet meer.

vincentweeda, 2020-04-13:
**Wat gebruik je?**
* DSMR-reader versie: v3.8
* Hardware: RaspberryPi 3B+
* Native of Docker *:  Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Sinds de laatste versie werkt het in en uitknijpen van de grafieken op de "Live graphs" niet meer op mijn Oneplus 5 (Android 9) toestel.

Ik heb het ook geprobeerd op de iPad iOS 13 maar daar functioneert het ook niet in Safari. 

Nu is het geen big deal maar wou het toch even melden.

ezweistra, 2020-04-14:
Ik heb deze bug inderdaad ook gemerkt op mijn Android toestellen (Samsung Galaxy Tab S3 en Samsung Galaxy S8, beide ook op Android 9). Werkte altijd perfect tot deze update.

dennissiemensma, 2020-04-14:
Bedankt voor jullie melding. Ik zal dan de library voor de grafieken in de volgende release weer downgraden. Wellicht zit er in die update iets wat niet lekker is.

dennissiemensma, 2020-05-02:
Opgelost voor volgende release

closed 2020-05-02 by dennissiemensma