dsmrreader/dsmr-reader #970: dsmr_datalogger

willemdej, 2020-04-15:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* dsmr-reader latest version*: v...
* Hardware *(RaspberryPi 4
* Native/docker..
* DSMR-protocol slimme meter *(v4)

Goedemorgen,

Na de installatie te hebben voltooid en een succes volle test meting te hebben gedaan met de P1 poort. Wil ik in de webbrowser de data uitlezen, heb inmiddels de goeie com poort ingevuld. (/dev/ttyUSB0) maar krijg in webbrowser de foutmelding; dsmr_datalogger niet geconfigureerd. De datalogger werkt succesvol in supervisor.

dennissiemensma, 2020-04-15:
Bedankt voor je melding. Het kan soms even duren voordat er gegevens getoond worden. Zijn ze nu wel zichtbaar? En zo nee, welke pagina bekijk je?

willemdej, 2020-04-20:
de gegevens zijn inmiddels zichtbaar, de DSMR-reader draait op een rpi ik heb nu een virtuele compoort aangemaakt met ser2net en kan de data uitlezen via Home Assistant. Ik krijg nu alleen in mijn sensors de totale gas/stroom consumptie. de energie productie wat ik per minuut verbruik werkt alleen niet.

dennissiemensma, 2020-04-20:
Ik begrijp niet helemaal wat je bedoelt met sensors. DSMR-reader kan alleen uitlezen en doorsturen wat je slimme meter aangeeft. De meter kan alleen productie weergeven als het niet in je huishouden opgenomen wordt (of het overschot ervan).

willemdej, 2020-04-21:
Had niet goed gekeken moet bij de ‘power’ sensors zijn. Werkt verder perfect 👍🏻

closed 2020-04-21 by dennissiemensma