dsmrreader/dsmr-reader #971: Huidig verbruik en huidig teruglevering

haijeploeg, 2020-04-15:
Hoi,

Ik heb een vraag met betrekking tot de interface van DSMR. Vandaag zijn er zonnepanelen geplaatst en er valt me direct iets op aan de interface. Ik zie gelijk dat ik energie terug lever. Maar ik zie dat mijn huidige verbruik op 0 staat. Klopt het dat als je energie opwekt je niet meer kan zien wat je huidige verbruik is?

Of zit hier een berekening in? Bijvoorbeeld ik wek 2500 watt op en verbruik nu 500 watt, dat de interface dit laat zien als dat ik op het moment 0 watt verbruik en 2000 watt terug lever?

Ben benieuwd hoe dit precies zit!

dennissiemensma, 2020-04-15:
Bedankt voor je vraag. DSMR-reader laat zien wat je slimme meter doorgeeft. Daar worden geen extra berekeningen qua teruglevering voor gebruikt.

Als je meer opwekt dan dat je verbruikt EN die elektriciteit kan terugvloeien in je huishouden (hangt af van je groepen etc), dan is het inderdaad mogelijk dat de slimme meter rapporteert dat je 0 Watt verbruikt.

Je kunt dit overigens ook zien op de slimme meter zelf. Als het constant 0 Watt is, zul je dat ook op de slimme meter moeten terugzien.

haijeploeg, 2020-04-16:
Ja je hebt gelijk. De slimme meter berekend dit automatisch al. Dus als ik 2500 watt opwek en 500 watt verbruik rapporteert de slimme meter 2000 watt teruglevering.

Moet je wel complimenten geven voor de app. Het werkt fantastisch!

dennissiemensma, 2020-04-16:
Fijn om te horen!

closed 2020-04-16 by dennissiemensma