dsmrreader/dsmr-reader #972: Netspanning toevoegen aan grafieken en MQTT

Jeltel, 2020-04-20:
Ik heb sinds kort zonnepanelen en die lees ik op een andere manier uit, Maar ik zou graag de netspanning weten. Dit omdat bij een te hoge netspanning de zonnepanelen uitgaan. Hiermee kan ik in de gaten houden of het risico op uitval aanwezig is.

Voor nu volstaat voor mij het toevoegen van de netspanning aan het MQTT topic, zodat ik deze in kan lezen in Home Asssistant. 
Mogelijk is het ook interessant om het toe te voegen aan een grafiek in dsmr zelf..

adebree, 2020-04-20:
Tenzij ik je niet begrijp, maar volgens mij is dit al  beschikbaar. In het `Telegram: Split topic` configuratie kan je een mapping maken zodat de voltages van de 3 fases in losse topics gepublished worden. De default mapping:

```
phase_voltage_l1 = dsmr/reading/phase_voltage_l1
phase_voltage_l2 = dsmr/reading/phase_voltage_l2
phase_voltage_l3 = dsmr/reading/phase_voltage_l3
```

Verder zie ik in de live graphs een 'recent phase voltages' de spanning van de fases over de afgelopen periode weergeven.

Jeltel, 2020-04-20:
Dan heb ik niet op de juiste plek gekeken. Ik zal nog eens beter mijn best doen!

closed 2020-04-20 by Jeltel