dsmrreader/dsmr-reader #973: DSMR-reader hangt na api call

casperleenheer, 2020-04-20:
**Wat gebruik je?**
* DSMR-reader versie v 3.8
* Hardware rPi 3b
* Native installatie
* DSMR-protocol slimme meter v5

Ik voer het volgende commando uit

curl  -H 'Authorization:Token MY-API-KEY'  http://192.168.x.x/api/v2/datalogger/dsmrreading

Daarna is dsmr-reader totaal unresponsive en moet ik de rPi rebooten. Daarna werkt alles weer als vanouds (goed dus).

dennissiemensma, 2020-04-20:
Bedankt voor je melding. Wat staat er in de logs? https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

Kun je op dat moment nog wel in het OS of werkt de hele Pi niet meer?

casperleenheer, 2020-04-20:
Ik kan nog wel in de rPi, alles is wel langzamer.

dennissiemensma, 2020-04-20:
Je kunt wellicht kijken met `top` welke processen er veel CPU of io in beslag nemen. En in de logsfiles kijken of daar iets vreemds aan de hand is.

casperleenheer, 2020-04-21:
Correctie, het dsmr komt wel weer terug in normale staat na de call, dus dat werkt gelukkig wel. Tijdens de call is het system niet vooruit te branden. Wanneer de call een timeout geeft zie je dit:
<html>
<head><title>504 Gateway Time-out</title></head>
<body bgcolor="white">
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.14.2</center>
</body>
</html>

In de dsmr_webinterface.log zie ik:

[2020-04-20 17:06:36 +0100] [2247] [INFO] Booting worker with pid: 2247
[2020-04-21 08:15:48 +0100] [611] [CRITICAL] WORKER TIMEOUT (pid:2247)
[2020-04-21 08:15:50 +0100] [10771] [INFO] Booting worker with pid: 10771

dennissiemensma, 2020-04-21:
Ik kan het helaas niet reproduceren hier.

- Zou je eens met `top` willen meekijken terwijl je het reproduceert? Ik ben dan benieuwd of een bepaald proces extra veel CPU pakt.

- Zou je ook willen kijken of je op de derde regel de waarde van `wa` (Waiting) dan direct ziet oplopen naar een hoge waarde? Screenshot, hier is die 0,1:

[image]

casperleenheer, 2020-04-22:
Voor de call

[image]

Tijdens call
[image]

[image]

Het qunicorn process is alleen zichtbaar tijdens de call.

dennissiemensma, 2020-04-22:
Bedankt voor je aanvulling. Het lijkt er duidelijk op dat die ergens op wacht, gezien `wa` op bijna 50% zit.

- Geef je trouwens de juiste of een ongeldige API auth-code op?

Hier staan dus geen fouten in verder? In deze logs:
- `/var/log/supervisor/dsmr_webinterface.log`
- `/home/dsmr/dsmr-reader/logs/django.log`
- `/home/dsmr/dsmr-reader/logs/dsmrreader.log`

Als daar ook niets in staat, kun je nog een stapje verder gaan, namelijk het proces zelf debuggen om te zien wat die doet. 
- Hier voor moet je even de applicatie niet open hebben in je browser, zodat het niet vervuilt.
- Je hebt het proces ID nodig en die wisselt constant:
```
sudo ps faux | grep dsmrreader.wsgi | grep -v grep
```

In dit voorbeeld is de PID `22097`:
```
dsmr       606  0.0  0.8  24464 16516 ?        S    Apr12   2:29  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
dsmr     22097  0.0  2.1  56840 42404 ?        S    Apr18   0:10  |   \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
```

- Vervolgens kun je dan `sudo strace -v -s 512 -f -p 22097` doen en in een andere tab je `curl` command. 

Als het goed is zie je een hoop voorbij regels komen. Wellicht hangt die ergens op of zie je iets anders. Het moet iets kleins zijn, maar ik ben er zelf niet tegenaan gelopen.

casperleenheer, 2020-04-23:
Dit is de dump, ik kan er geen chocola van maken:

`
strace: Process 26720 attached
_newselect(8, [5 7], [], [], {tv_sec=18, tv_usec=489908}) = 1 (in [5], left {tv_sec=14, tv_usec=854230})
fchmod(6, 001)                          = 0
accept4(5, {sa_family=AF_UNIX}, [110->2], SOCK_CLOEXEC) = 11
ioctl(11, FIONBIO, [0])                 = 0
fcntl64(11, F_GETFD)                    = 0x1 (flags FD_CLOEXEC)
fcntl64(11, F_SETFD, FD_CLOEXEC)        = 0
recv(11, "GET /api/v2/datalogger/dsmrreading HTTP/1.0\r\nX-Forwarded-For: 192.168.3.185\r\nHost: 192.168.3.223\r\nConnection: close\r\nUser-Agent: curl/7.64.0\r\nAccept: */*\r\nAuthorization: Token J36EQIZI4NCBF1NJNC4WL44EL3PVNRDA3XMB18BAW7YCELGXJTJZT9P2CL6LOFQJ\r\n\r\n", 8192, 0) = 244
getpeername(11, {sa_family=AF_UNIX}, [110->2]) = 0
getsockname(5, {sa_family=AF_UNIX, sun_path="/var/tmp/gunicorn--dsmr_webinterface.socket"}, [110->46]) = 0
send(15, "\27\3\3\0\26^g\205o\260\266\321&\205\266\245\307\336\205]\234hM\25\2167\334", 27, MSG_NOSIGNAL) = 27
send(15, "\27\3\3\0\23\327\323\324\t;i\357\25\26\363\10W\315\312EI\235\16\251", 24, MSG_NOSIGNAL) = 24
close(15)                               = 0
getpid()                                = 26720
getpid()                                = 26720
stat64("/etc/resolv.conf", {st_dev=makedev(0xb3, 0x2), st_ino=962, st_mode=S_IFREG|0644, st_nlink=1, st_uid=0, st_gid=0, st_blksize=4096, st_blocks=8, st_size=49, st_atime=1569457463 /* 2019-09-26T01:24:23.128832774+0100 */, st_atime_nsec=128832774, st_mtime=1587567393 /* 2020-04-22T15:56:33.447043069+0100 */, st_mtime_nsec=447043069, st_ctime=1587567393 /* 2020-04-22T15:56:33.447043069+0100 */, st_ctime_nsec=447043069}) = 0
openat(AT_FDCWD, "/etc/hosts", O_RDONLY|O_CLOEXEC) = 15
_llseek(15, 0, [0], SEEK_CUR)           = 0
fstat64(15, {st_dev=makedev(0xb3, 0x2), st_ino=909, st_mode=S_IFREG|0644, st_nlink=1, st_uid=0, st_gid=0, st_blksize=4096, st_blocks=8, st_size=131, st_atime=1569457441 /* 2019-09-26T01:24:01.260437241+0100 */, st_atime_nsec=260437241, st_mtime=1569456598 /* 2019-09-26T01:09:58.729161048+0100 */, st_mtime_nsec=729161048, st_ctime=1569457441 /* 2019-09-26T01:24:01.260437241+0100 */, st_ctime_nsec=260437241}) = 0
read(15, "127.0.0.1\tlocalhost\n::1\t\tlocalhost ip6-localhost ip6-loopback\nff02::1\t\tip6-allnodes\nff02::2\t\tip6-allrouters\n\n127.0.1.1\traspberrypi\n", 4096) = 131
_llseek(15, 0, [131], SEEK_CUR)         = 0
read(15, "", 4096)                      = 0
close(15)                               = 0
socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_ROUTE) = 15
bind(15, {sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, 12) = 0
getsockname(15, {sa_family=AF_NETLINK, nl_pid=26720, nl_groups=00000000}, [12]) = 0
sendto(15, {{len=20, type=0x16 /* NLMSG_??? */, flags=NLM_F_REQUEST|0x300, seq=1587624299, pid=0}, "\x00\x00\x00\x00"}, 20, 0, {sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, 12) = 20
recvmsg(15, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, msg_namelen=12, msg_iov=[{iov_base=[{{len=76, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, "\x02\x08\x80\xfe\x01\x00\x00\x00\x08\x00\x01\x00\x7f\x00\x00\x01\x08\x00\x02\x00\x7f\x00\x00\x01\x07\x00\x03\x00\x6c\x6f\x00\x00\x08\x00\x08\x00\x80\x00\x00\x00\x14\x00\x06\x00\xff\xff\xff\xff\xff\xff\xff\xff\xca\x00\x00\x00\xca\x00\x00\x00"}, {{len=88, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, "\x02\x18\x00\x00\x02\x00\x00\x00\x08\x00\x01\x00\xc0\xa8\x03\xdf\x08\x00\x02\x00\xc0\xa8\x03\xdf\x08\x00\x04\x00\xc0\xa8\x03\xff\x09\x00\x03\x00\x65\x74\x68\x30\x00\x00\x00\x00\x08\x00\x08\x00\x00\x02\x00\x00\x14\x00\x06\x00\x80\xe0\x00\x00\xf6\x1b\x01\x00\x94\xc2\xa8\x00\x95\xad\xea\x00"}], iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 164
recvmsg(15, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, msg_namelen=12, msg_iov=[{iov_base=[{{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, "\x0a\x80\x80\xfe\x01\x00\x00\x00\x14\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x14\x00\x06\x00\xff\xff\xff\xff\xff\xff\xff\xff\xca\x00\x00\x00\xca\x00\x00\x00\x08\x00\x08\x00\x80\x00\x00\x00"}, {{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, "\x0a\x40\x00\x00\x02\x00\x00\x00\x14\x00\x01\x00\xfd\xc1\xc9\xbd\xfc\xbc\x00\x01\xde\xee\xd4\xde\x60\xdc\x1b\xad\x14\x00\x06\x00\x0b\x1c\x00\x00\x0b\x1c\x00\x00\xab\xe3\xa8\x00\x74\x8f\xff\x00\x08\x00\x08\x00\x00\x03\x00\x00"}, {{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, "\x0a\x40\x80\xfd\x02\x00\x00\x00\x14\x00\x01\x00\xfe\x80\x00\x00\x00\x00\x00\x00\x92\x13\x92\xd3\x4d\x71\x83\xdd\x14\x00\x06\x00\xff\xff\xff\xff\xff\xff\xff\xff\x23\xc0\xa8\x00\x23\xc0\xa8\x00\x08\x00\x08\x00\x80\x00\x00\x00"}], iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 216
recvmsg(15, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, msg_namelen=12, msg_iov=[{iov_base={{len=20, type=NLMSG_DONE, flags=NLM_F_MULTI, seq=1587624299, pid=26720}, 0}, iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 20
close(15)                               = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_IP) = 15
connect(15, {sa_family=AF_INET, sin_port=htons(5432), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
getsockname(15, {sa_family=AF_INET, sin_port=htons(41914), sin_addr=inet_addr("127.0.0.1")}, [28->16]) = 0
[... cut ...] 6\320\2550\317\216\350\r_\251\34\224in", 91, MSG_NOSIGNAL) = 91
poll([{fd=15, events=POLLIN|POLLERR}], 1, -1) = ? ERESTART_RESTARTBLOCK (Interrupted by signal)
--- SIGABRT {si_signo=SIGABRT, si_code=SI_USER, si_pid=532, si_uid=1001} ---
getpid()                                = 26720
write(8, "\6", 1)                       = 1
sigreturn({mask=[]})                    = -1 EINTR (Interrupted system call)
poll([{fd=15, events=POLLIN|POLLERR}], 1, -1 <unfinished ...>) = ?
+++ killed by SIGKILL +++
`

dennissiemensma, 2020-04-23:
Bedankt voor de gegevens. Ik neem aan dat DSMR-reader verder wel goed bij je draait qua webinterface toch?

Het enige wat ik uit die dump kan halen is dat ik een paar keer `Resource temporarily unavailable` zie, maar daar staat verder ook niets bij.

Ik vind het verder lastig om te zeggen wat het nog meer kan zijn. Ik kan me ook niet herinneren dat iemand een soortgelijk iets heeft gemeld. Dat wijst wel vaak op een lokaal probleem, maar daar ben je natuurlijk niet mee geholpen.

Je zou nog eens kunnen kijken wat er gebeurt als je een foutieve API-key meegeeft. Of simpelweg een ander endpoint proberen, of dat hetzelfde issue geeft.

casperleenheer, 2020-04-24:
Ok, interessante ontwikkelingen:

- /api/v2/consumption/gas -> werkt

- /api/v2/consumption/electricity -> werkt

- /api/v2/datalogger/dsmrreading -> werkt niet

dennissiemensma, 2020-04-24:
Wil je eens kijken of er wel indexes staan op die tabel? Als dat niet zo is dan verklaart dat de toename aan I/O via `wa` en ook dat het te lang duurt.

```
sudo su - postgres
psql dsmrreader

\dS dsmr_datalogger_dsmrreading;
```

Onderaan zou je zoiets moeten zien:
```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

casperleenheer, 2020-04-24:
Die heb ik ook, alleen niet 1 met die _uniq.

Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_processed_add8a852" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" btree ("timestamp")

Wil je eens kijken of er wel indexes staan op die tabel? Als dat niet zo is

dennissiemensma, 2020-04-24:
Hoeveel records heb je uberhaupt in die tabel staan?

```
select count(id) from dsmr_datalogger_dsmrreading;
```

casperleenheer, 2020-04-24:
10847815

Daar deed ie meer dan een minuut over....

Waar het mij allemaal om ging was het voltage van mijn drie fasen te
krijgen via mqtt. Als ik
Change (Data source) Telegram: Split topic aanzet, krijg ik 1 of meer
messages per seconde op mijn mqtt broker binnen. Ik gebruik Home Assistant
en die ging daarop ook los. Technisch werkte het, maar gewoon te veel
berichten waardoor de sqlite database van HA er op een gegeven moment er
ook uit klapte. Ik heb die optie dus maar weer snel uitgezet. Ik probeer
dus via de api de waarden uit te lezen. Twee tickets later... :-)

Maar goed, het lijkt mij dat deze tabel dus opgeschoond moet worden. Nu ben
ik wel handig met Sql Server, maar Postgres is nieuw voor mij, dus als je
mij daarmee wil helpen, zou dat top zijn.

dennissiemensma, 2020-04-24:
* In de admin kun je hier het roteren van gegevens aanzetten: `/admin/dsmr_datalogger/retentionsettings/`

Dat zal wel even duren de eerste keer, maar die opschoning blijft wel aan de gang vermoed ik.

* Daarnaast is het handig om, gezien je een v5-meter hebt, het aantal metingen per minuut te limiteren. Je krijgt er zo'n 60, maar 10 of minder is ook prima. Je kunt de datalogger sleep op bijvoorbeeld 5 of zelfs 10 seconden zetten in de admin: `/admin/dsmr_datalogger/dataloggersettings/`

Hiermee krijg je dus ook minder data binnen, slaat het systeem minder op en stuurt die minder vaak MQTT uit. Voor de nauwkeurigheid en de dagtotalen maakt dat niets uit overigens.

casperleenheer, 2020-04-28:
Als je de retentie van je data op een week zet, gaat het goed.

closed 2020-04-28 by casperleenheer