dsmrreader/dsmr-reader #974: Dag rapporten worden niet meer gegenereerd

geforceFXguy, 2020-04-20:
Sinds 4 april worden mijn dag reports niet meer gemaakt. Nu heb ik de logging aangezet en dan kom hij met de melding dat er geen gegevens zijn voor 5 april. Nu heb ik gekeken en in de database staat toch echt gegevens in. 

Zit op V3.8
[image]

[image]

[image]

[image]

[image]

dennissiemensma, 2020-04-20:
Bedankt voor je melding en de uitgebreide info. Gebruik je toevallig MySQL als database-opslag? (#909)

geforceFXguy, 2020-04-20:
ik gebruik:
MySQL 5.7.25 deze draait op een Windows 2016 machine

dennissiemensma, 2020-04-20:
Dan vermoed ik dat het mogelijk ligt aan missende tijdzone definities. Deze staan standaard niet aan in MySQL en zorgen voor vage bugs. 

Zie ook het topic dat ik hierboven al heb gelinked: https://github.com/dennissiemensma/dsmr-reader/issues/909#issuecomment-594081194

Ik verwacht dat dat de problemen oplost.

geforceFXguy, 2020-04-20:
Het lijkt nu weer te werken. Apart dat het van vanaf 2017 al gewerkt heeft :)

dennissiemensma, 2020-04-20:
Fijn om te horen. Het kan zijn dat er in recentere versies een edge-case is dit veroorzaakt. Hoe dan ook wordt dit project niet gemaakt en getest zonder tijdzone-support, dus vandaar dat de gerelateerde issues daaruit vaak wat vaag zijn.

closed 2020-04-20 by dennissiemensma