dsmrreader/dsmr-reader #975: dsmr 3.8 configuration menu not working

marco2152, 2020-04-24:
I noticed that I cannot use the Configuration menu anymore. I got this error page when opening the Configuration. I tried updating, but as 3.8 is the latest version, this didn't solve the issue:
Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: column dsmr_backup_backupsettings.compression_level does not exist
LINE 1: ...kup_time", "dsmr_backup_backupsettings"."folder", "dsmr_back...
                                                             ^
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/mixins.py", line 52, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/configuration.py", line 31, in get_context_data
    backup_settings=BackupSettings.get_solo(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 559, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 411, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 258, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1261, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 57, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1151, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 68, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 77, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)

dennissiemensma, 2020-04-24:
Hi, can you try whether this resolves the issue: https://github.com/dennissiemensma/dsmr-reader/issues/886#issuecomment-587922224

marco2152, 2020-04-24:
> Hi, can you try whether this resolves the issue: #886 (comment)

Ja, dat heeft geholpen. Bedankt!

Groeten,
Marco

closed 2020-04-24 by dennissiemensma