dsmrreader/dsmr-reader #977: supervisor.sock issue

GreyEarl, 2020-04-26:
**DSMR-reader versie** 
v3.8
**Hardware**
Intel NUC (x64)
**Native of Docker**
Native
**DSMR-protocol**
V5

**Issue**

Zodra ik mijn systeem reboot start DSMR niet meer op vanwege issues met Supervisor.
```
$ sudo supervisorctl
unix:///var/run/supervisor.sock no such file
```
Na lang zoeken heb ik een tijdelijk workaround:

```
sudo touch /var/run/supervisor.sock
sudo supervisord -c /etc/supervisor/supervisord.conf
supervisorctl restart all
```

Dit werkt tot de volgende reboot. Ik heb Supervisor verwijderd en opnieuw geïnstalleerd, geen succes. Ik heb op een gegeven moment heel DSMR verwijderd en opnieuw opgezet, het probleem blijft hetzelfde,

Het probleem met Supervisor is begonnen toen ik ben gaan upgraden naar v3.x. Iemand enig ervaring hiermee?

dennissiemensma, 2020-04-27:
Heb je ook al in de logfiles van Supervisor gekeken? Ik zou verwachten dat daar meer informatie staat:

`/var/log/supervisor/`

Heb je ook een nieuwere Python-versie moeten installeren voor de upgrade naar 3.x?

GreyEarl, 2020-04-28:
Yes, maar het zegt naar mijn mening niet veel. Dit is het moment vanaf een reboot:

```
2020-04-28 22:54:32,548 INFO exited: dsmr_client_datalogger (terminated by SIGTERM; not expected)
2020-04-28 22:54:32,555 WARN received SIGHUP indicating restart request
2020-04-28 22:54:32,556 INFO waiting for dsmr_webinterface, dsmr_backend to die
2020-04-28 22:54:32,561 WARN received SIGTERM indicating exit request
2020-04-28 22:54:33,619 INFO stopped: dsmr_backend (exit status 0)
2020-04-28 22:54:35,622 INFO waiting for dsmr_webinterface to die
2020-04-28 22:54:38,627 INFO waiting for dsmr_webinterface to die
2020-04-28 22:54:41,632 INFO waiting for dsmr_webinterface to die
2020-04-28 22:54:44,636 WARN killing 'dsmr_webinterface' (5814) with SIGKILL
2020-04-28 22:54:44,636 INFO waiting for dsmr_webinterface to die
2020-04-28 22:54:44,638 INFO stopped: dsmr_webinterface (terminated by SIGKILL)
```

Er gebeurt verder niets totdat ik de workaround uitvoer, waarna ik het volgende zie staan:

```
2020-04-28 23:00:15,716 CRIT Supervisor running as root (no user in config file)
2020-04-28 23:00:15,716 INFO Included extra file "/etc/supervisor/conf.d/dsmr_backend.conf" during parsing
2020-04-28 23:00:15,716 INFO Included extra file "/etc/supervisor/conf.d/dsmr_client_datalogger.conf" during parsing
2020-04-28 23:00:15,716 INFO Included extra file "/etc/supervisor/conf.d/dsmr_datalogger.conf" during parsing
2020-04-28 23:00:15,716 INFO Included extra file "/etc/supervisor/conf.d/dsmr_webinterface.conf" during parsing
2020-04-28 23:00:16,052 INFO RPC interface 'supervisor' initialized
2020-04-28 23:00:16,052 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-04-28 23:00:16,053 INFO daemonizing the supervisord process
2020-04-28 23:00:16,054 INFO supervisord started with pid 3293
2020-04-28 23:00:17,059 INFO spawned: 'dsmr_webinterface' with pid 3298
2020-04-28 23:00:17,062 INFO spawned: 'dsmr_client_datalogger' with pid 3299
2020-04-28 23:00:17,064 INFO spawned: 'dsmr_backend' with pid 3300
2020-04-28 23:00:18,894 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-04-28 23:00:20,971 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2020-04-28 23:00:20,971 INFO success: dsmr_client_datalogger entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2020-04-28 23:00:24,681 INFO waiting for dsmr_webinterface to stop
2020-04-28 23:00:24,682 INFO waiting for dsmr_client_datalogger to stop
2020-04-28 23:00:24,682 INFO waiting for dsmr_backend to stop
2020-04-28 23:00:24,683 INFO stopped: dsmr_client_datalogger (terminated by SIGTERM)
2020-04-28 23:00:25,832 INFO stopped: dsmr_webinterface (exit status 0)
2020-04-28 23:00:25,854 INFO stopped: dsmr_backend (exit status 0)
2020-04-28 23:00:25,862 INFO spawned: 'dsmr_webinterface' with pid 3315
2020-04-28 23:00:25,865 INFO spawned: 'dsmr_client_datalogger' with pid 3316
2020-04-28 23:00:25,868 INFO spawned: 'dsmr_backend' with pid 3317
2020-04-28 23:00:27,219 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-04-28 23:00:28,896 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2020-04-28 23:00:28,896 INFO success: dsmr_client_datalogger entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
```
Ik durf het niet met zekerheid meer te zeggen, maar volgens mij had ik inderdaad een nieuwere Python versie moeten installeren.

dennissiemensma, 2020-04-28:
Wat krijg je hier te zien:

```
sudo su - dsmr
python3 --version
```

GreyEarl, 2020-04-28:
Als volgt:
```
dsmr@INTEL-NUC:~$ python3 --version
Python 3.7.5
```

dennissiemensma, 2020-04-30:
Je Python-versie is prima. Wat voor OS draai je op je machine? Is dat een Debian-like distro?

Normaal gesproken zou het (her)installeren van Supervisor ervoor moeten zorgen dat het systeem bij booten ervoor zorgt dat alles goed staat. Echter, wellicht is het hier iets kleins qua permissies of iets, wat opzich wel raar is gezien het als root-user zou moeten draaien (het hoofdproces).

Je zou nog kunnen proberen om je fix qua `sudo touch /var/run/supervisor.sock` op te nemen in een `@reboot` trigger. Mits dat hetgene is wat het veroorzaakt.

Een ander ding is kijken of er een systemctl job voor is. Ik ben er niet helemaal thuis in voor alle OS'en, maar op Raspbian hier zie ik bijvoorbeeld:
```
# /etc/systemd/system/multi-user.target.wants/supervisor.service

[Unit]
Description=Supervisor process control system for UNIX
Documentation=http://supervisord.org
After=network.target

[Service]
ExecStart=/usr/bin/supervisord -n -c /etc/supervisor/supervisord.conf
ExecStop=/usr/bin/supervisorctl $OPTIONS shutdown
ExecReload=/usr/bin/supervisorctl -c /etc/supervisor/supervisord.conf $OPTIONS reload
KillMode=process
Restart=on-failure
RestartSec=50s

[Install]
WantedBy=multi-user.target
```

Er kan ook zoiets zijn voor jouw OS wat misschien niet lekker gaat. Het moet iets kleins zijn, gezien je aangeeft dat het eerder wel heeft gewerkt.

Een ander ding wat je nog kunt proberen is om het pad `/var/run/supervisor.sock` in de Supervisor config aan te passen naar een andere map zoals `/var/tmp/supervisor.sock`.

GreyEarl, 2020-05-03:
Hi Dennis,

Dank voor je reply. Ik zit op Ubuntu 18.04. Ik heb tevergeefs al eerder een ander pad geprobeerd in supervisor.sock. Ik heb supervisor.service op de juiste plek staan. Ik denk nu dat er iets mis is met deze service. Het vreemde is dat DSMR gewoon werkt.

```
$ sudo systemctl status supervisord.service
● supervisord.service - Supervisor daemon
   Loaded: loaded (/etc/systemd/system/supervisord.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2020-05-02 20:22:55 CEST; 19s ago
     Docs: http://supervisord.org
  Process: 13435 ExecStart=/usr/bin/supervisord -n -c /etc/supervisor/supervisord.conf (code=exited, status=1/FAILURE)
 Main PID: 13435 (code=exited, status=1/FAILURE)

May 02 20:22:55 INTEL-NUC systemd[1]: supervisord.service: Main process exited, code=exited, status=1/FAILURE
May 02 20:22:55 INTEL-NUC systemd[1]: supervisord.service: Failed with result 'exit-code'.
```
Ik ga je @reboot trigger suggestie uitproberen, anders probeer ik de docker versie te gebruiken. Feit is dat er eigenlijk niets mis lijkt te zijn met de DSMR applicatie :) Thanks voor je support!

closed 2020-05-03 by GreyEarl