dsmrreader/dsmr-reader #978: Upgrade ubuntu 19.10 naar 20.4: No module named Django

brainfrz82, 2020-04-30:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.8
* Hardware *(RaspberryPi 3, 4 of iets anders)*:  Intel NUC 
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v4

Hoi,

Na upgrade van ubuntu 19.10 naar 20.4 wou helaas mijn DSMR-Reader niet meer starten. Inloggen op de virtualenv gaf al snel de foutmelding "No module named Django". Installeren met Pip3 gaf ook foutmeldingen No module named 'pip._vendor.packaging.tags'.

Uiteindelijk besloten om de virtualenv maar opnieuw te maken door het volgende stuk opnieuw uit te voeren:
```
#oude virtualenv weghalen
mv /home/dsmr/.virtualenvs /home/dsmr/.virtualenvs.old

# Virtual env
sudo sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --python python3
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'

# Config & requirements
sudo sudo -u dsmr cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/django/postgresql.py /home/dsmr/dsmr-reader/dsmrreader/settings.py
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

# Setup
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py collectstatic --noinput
```
Daarna kon ik DSMR-Reader weer goed gebruiken, alles lijkt in orde. Nu heb ik 2 vragen:

-Heb ik daarmee misschien onder water nog wat stukgemaakt wat ik nu niet zie?
-het virtualenv commando (versie 20.0.17) wat onder ubuntu 20.4 meegeleverd wordt snapt de parameter "--no-site-packages " niet. Ik heb deze weggelaten en het lijkt geen problemen te geven, maar is dat ook zo?

Waarschijnlijk zijn er efficiëntere fixes, maar als heads-up voor mensen die upgraden, doe er je voordeel mee :)

dennissiemensma, 2020-04-30:
Bedankt voor het delen van je bevindingen. Doorgaans is het bij een distro-upgrade afdoende om de virtualenv opnieuw aan te maken. Voor de rest zijn er geen dependencies. 

Die parameter die mist of uitgefaseerd is, is niet erg. Ik heb die altijd gebruikt om legacy-redenen, dus waarschijnlijk is die dan niet meer nodig.

Wat betreft dingen die je niet ziet: Als de applicatie verder start en data opslaat/verwerkt, dan verwacht ik geen issues. Ik kan je wel aanraden om te kijken of het maken van database-backups nog goed werkt, dus of de backups bestanden zijn met queries en daarin data.

brainfrz82, 2020-05-01:
Dank je voor je antwoord, de backup ziet er ook nog goed uit. Ik ga dr inderdaad van uit dat dit verder geen problemen geeft :)

closed 2020-05-01 by dennissiemensma