dsmrreader/dsmr-reader #979: Verbruik tussen 2 tijdstippen

eivcka38aDa3aq, 2020-05-01:
Ik ben erg tevreden over DSMR. Ontzettend bedankt voor het maken en beheren. 

Ik heb één sugggestie die DSMR-reader voor mij nog beter zou maken. Het zou voor mij namelijk erg handig zijn als ik het verbruik tussen 2 tijdstippen kan berekenen. Het meest handig hiervoor lijkt mij om dit op te nemen in de live grafiek. Als ik mijn live grafiek bijvoorbeeld het vebruik tussen 13u en 15u laat zien zou DSMR-reader het vebruik in die periode kunnen bepalen. Op deze manier kan ik accurater achterhalen wat mijn wasmachine, koelkast etc. verbruikt. Uiteraard speelt het sluipverbruik hierbij ook een rol, maar het zou mij erg helpen om dit nauwkeuriger in kaart te brengen.

dennissiemensma, 2020-05-01:
Bedankt voor je suggestie. Ik weet niet of dat mogelijk is binnen de grafieken-tool die ik gebruik, maar ik zal er tzt een keer naar kijken.

dennissiemensma, 2021-04-18:
Samenvoeging uit ander issue #1362

> Ik zou graag een selectie in de 'live' grafiek willen maken, en de totale verbruik+kosten van die selectie op basis van mijn huidige contract willen kunnen zien.
> 
> Dus in bijvoorbeeld de 'Recent electricity usage' grafiek, mijn muis gebruiken om een blok te selecteren, en gelijk in een popup totaal verbruik + kosten voor die selectie zien.
> 
> Dat maakt het direct inzichtelijk wat bepaalde activiteiten kosten. Bijvoorbeeld, als ik de ovenreiniging aanzet, en die 2u20m draait - voor een deel in hoog tarief en een deel in laag tarief - zou het mooi zijn als ik in de grafiek die periode van 2u20m kon selecteren, en de verbruik+kosten automatisch voor mij berekend werden om te zien wat die ovenreiniging mij gekost heeft.

dennissiemensma, 2022-07-08:
Dit bleek uiteindelijk te complex om te maken helaas. Wellicht dat ik in de plaats daarvan ooit een cumulatieve grafiek toevoeg. Dan kun je vrij makkelijk zien wat het kWh gebruik op punt X is en op punt Y. 

Nu is dat uberhaupt niet te zien omdat er geen kWh-grafiek is (alleen achteraf in het Archief, gegroepeerd op uur).

dennissiemensma, 2022-07-13:
In v5.5 komt wat rework van de Live graphs.

- Negatieve grafieken zijn nu weer weg, zoals vanouds
- Grafieken hebben nu een muishover met een kruis dat iets nauwkeurigere informatie geeft
- Het kruis is vooral handig bij de nieuwe toevoeging aan de Watt-grafiek, nu staat er ook de relatieve kWh in

Het is nog niet helemaal perfect, omdat het relatief is aan het eerste datapunt dat getoond wordt, maar het zegt opzich genoeg en je kunt daarmee goed zien wat het verbruik is tussen twee punten:

## Screenshot
Het is nog wel wat handwerk om te rekenen, maar ik denk dat het een goed begin is.

Voor onderstaand voorbeeld heb ik de muis helemaal rechts op het einde rond `21:01`, maar horizontaal ten hoogte van ~`16:00`. Op dat tijdstip stond het verbruik op `8,03` kWh en het verticale punt staat op `9,403` kWh, dus ik heb in die tijdsspanne zo'n `1,4` kWh verbruikt.

[image]

dennissiemensma, 2022-07-13:
Zie bovenstaande @chasewinnings391 / @gotiniens / @joopy / @simkin

dennissiemensma, 2022-07-13:
In latere releases kan ik het nog wat finetunen. Het liefste houd ik het 0-punt van de kWh gewoon op whatever datapunt links *zichtbaar* is, ipv het eerste datapunt wat teruggegeven wordt adhv de gebruikersinstellingen qua grafiekenperiode.

Een alternatief is uitzoeken of ik met eCharts iets maken om met de muis een begin/eind te selecteren en dat dan het kWh-verschil berekend wordt. Dat zou ook werkbaar zijn.

closed 2022-07-13 by dennissiemensma

dennissiemensma, 2022-07-15:
Uitgebracht in [v5.5](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.5.0)

chasewinnings391, 2022-07-15:
Super! Erg mooie oplossing zo! Bedankt @dennissiemensma

Mixpower, 2022-07-17:
> Uitgebracht in [v5.5](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.5.0)

Ik vind dit geen fijne toevoeging, ik vond de oude implementatie overzichtelijker is dit met een instelling te wijzigen?

dennissiemensma, 2022-07-17:
Bedankt voor je feedback! Ik denk dat je het beste kunt blijven of downgraden op de versie die voor jou het prettigst is. DSMR-reader updates zijn doorgaans niet noodzakelijk voor de werking van oude versies, tenzij vermeld in de changelog (zelden). Het is namelijk onwerkbaar voor mij om de interface te updaten of iets te proberen en tevens alle mogelijke (oude) varianten te blijven ondersteunen.

uitje, 2022-07-19:
Is het ook mogelijk om de live grafieken altijd alleen de data van de huidige dag mee te laten nemen? Op deze manier geven de totalen de meest nuttige info weer in de grafieken.

dennissiemensma, 2022-07-19:
@uitje bedankt voor je suggestie. Ik heb er al wat mee gespeeld tijdens het bouwen, maar de live grafieken lenen zich daar minder goed voor, omdat het de afgelopen X periode is, die per gebruiker anders is. Het kan het laatste uur zijn of de laatste week.

Wellicht dat ik nog iets kan vinden waardoor er met de muis een selectie te maken is om het verschil te tonen. Dan kan ik namelijk gewoon de meterstanden daar tonen, die komen het dichtste bij de werkelijkheid, maar zijn vanzelfsprekend wat kryptischer zonder goede diff.

uitje, 2022-07-20:
Ja dat snap ik. Zou het wel mogelijk zijn om in de configuratie een optie mee te nemen waarmee je het weergeven van de totalen in de live grafieken kan aan/uit zetten. De totalen geven nu niet echt nuttige info weer en de schaal van de grafieken wordt hier ook op aangepast. Ik heb de grafieken altijd op m'n oude iPad aan staan ;)

dennissiemensma, 2022-07-20:
Ik denk dat dat vrij onwerkbaar wordt. Voor een uitgebreid verhaal zie #1673.

Mijn advies is om voor nu te downgraden naar de vorige versue als het je echt in de weg zit. Elke update brengt veranderingen mee, maar updates zijn zelden noodzakelijk.

dennissiemensma, 2022-08-02:
Onderdeel van [v5.6](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.6.0)