dsmrreader/dsmr-reader #980: Afwijkend verbruik tegenover opmeting

braham2019, 2020-05-04:
**Wat gebruik je?**
versie 3.8 op Hardware RaspberryPi 4, Native 
slimme meter v5, ingesteld voor België Fluvius

Beste Dennis,

Als ik de totalen van de energieleverancier (die automatisch de informatie krijgt van de netbeheerder) neem voor de maand april, kom ik op het volgende:
Verbruik piek 132,8  Kwh
verbruik dal 122,7
Productie piek 114,3
productie dal 31,3

Als ik de totalen neem die in DSMR applicatie staan, kom ik op
Verbruik piek  93 Kwh
verbruik dal 131
Productie piek 218
productie dal 73

Waar kan dit verschil vandaan komen ?

closed 2020-05-04 by braham2019

dennissiemensma, 2020-05-04:
Ik neem aan dat je deze zelf bewust weer gesloten had? Of klikte je verkeerd?

braham2019, 2020-05-04:
Ik heb deze zelf gesloten omdat ik merkte, na extra nazicht, dat aan de cijfers van mijn energie leverancier eea serieus verkeerd was. 

Ik dacht misschien dat jouw applicatie eea verkeerd aan het uitlezen was, maar dat is niet zo. Waarvoor dank :-)