dsmrreader/dsmr-reader #981: Hulp bij esp8266 en dsmr-reader

erikkral, 2020-05-04:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v...de laatste
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi ... esp8266 met espeasy
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: ...docker
* DSMR-protocol slimme meter *(v4 of v5)*: v...

Hallo,

Ik weet niet zeker of ik hier goed zit of dat dit hoort bij een bestaand issue.
Ik ben zoekende naar een manier om mijn slimme meter-gegevens van een ESP8266 (NodeMCU) met espeasy in dsmr-reader te krijgen. 
In Domoticz lukt dat zonder al te veel moeite. 

Echter bij dsrm-reader loop ik vast omdat niet meer weet hoe ik het moet doen. Ik ervaar de documentatie voor wat betreft API en MQTT als summier. 

Wordt er ergens beschreven hoe ik gegevens van een ESP8266 gemakkelijk in DSMR-reader kan krijgen? Ik heb iets gelezen over een MQTT broker zoals Mosquitto. Blijkbaar kan je toch met MQTT gegevens importeren in dsrm-reader terwijl de MQTT helppagina zegt dat je het alleen kan gebruiken om gegevens te exporteren vanuit dsrm-reader. ??

 Als iemand mij kan helpen, dan wil ik wel aanbieden om dit op papier te zetten zodat het opgenomen kan worden in de handleiding.

dennissiemensma, 2020-05-04:
Bedankt voor je vraag. Hierbij wat informatie voor je:
- DSMR-reader ondersteunt alleen **uitgaande** MQTT-berichten naar een broker.
- DSMR-reader kan alleen telegrammen **binnenkrijgen** via de ingebouwde datalogger, de v1 API (remote datalogger) of een handmatig script (v2 API).
- De documentatie voor de (v1/v2) API is een aantal releases terug verplaatst van ReadTheDocs naar je lokale installatie van DSMR-reader op `/docs/api`. Daar staan ook voorbeelden, inclusief een complete Postman-collectie om je te helpen.

mosjonathan, 2020-05-05:
hallo erik,
kan denk ik wel je vraag van een antwoord voorzien.
even voor mijn beeld je hebt de esp8266 aan de slimme meter hangen?
en op de pi docker?
zo ja kan je onderstaand uitvoeren.

begin met installeren van socat op je rpi

sudo apt install socat
maak in /etc/systemd/system het volgende bestand aan:  socat.service

zet de volgende inhoud in dit bestand:
`[Unit]
Description=Socat

[Service]
Type=simple
ExecStart=/usr/bin/socat -U pty,link=/dev/virtualcom0,raw tcp:(IP_adress_van_ESP8266):PoortESP&
User=root
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target`

hierna sudo systemctl reload-daemon
daarna start je de service met het volgende commando: 
sudo systemctl start socat.service

nu kun je checken of je telegrammen voorbij ziet komen met het volgende commando:
sudo cat /dev/virtualcom0

als het werkt kun je met het volgende commando zorgen dat de service mee start als de pi boot:
sudo systemctl enable socat.service

als laatste moet je op de pi in de config van de container /dev mounten als volume:
    volumes:
      - /dev/:/dev/
hierna kun je in de webinterface van dsmr reader virualcom0 kiezen als bron.

erikkral, 2020-05-14:
Dag allen,

Bedankt voor jullie hulp. Inmiddels is het mij gelukt door gebruik te maken van het Python script van ThinkpadNL.

Dat script heb ik nu draaien op dezelfde server als dsmr reader. 

Dit issue mag gesloten worden.

closed 2020-05-14 by erikkral