dsmrreader/dsmr-reader #984: Decimaal phase_voltage_ in MQTT

TopdRob, 2020-05-06:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v.3.91
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 2B+
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

De voltagewaarden hebben 1 decimaal maar in de MQTT berichten Telegram (zowel geslitst als json) is het afgerond op 0 decimaal.
phase_voltage_l1 = dsmr/reading/phase_voltage_l1
phase_voltage_l2 = dsmr/reading/phase_voltage_l2
phase_voltage_l3 = dsmr/reading/phase_voltage_l3

dennissiemensma, 2020-05-07:
Bedankt voor je melding. Ik kan het helaas niet reproduceren. 

Als ik het met een test probeer krijg ik in split topic telegram:
```
dsmr/telegram/id 13537089
dsmr/reading/phase_voltage_l3 229.2
dsmr/reading/phase_voltage_l2 228.1
dsmr/reading/phase_voltage_l1 232.0

dsmr/telegram/id 13537088
dsmr/reading/phase_voltage_l3 230.2
dsmr/reading/phase_voltage_l2 230.1
dsmr/reading/phase_voltage_l1 229.0
```

Begrijp ik het goed dat je in de grafieken enzo wel decimalen ziet?

TopdRob, 2020-05-07:
Dit zijn de waarden die ik krijg:

```
dsmr/telegram/id 38860469
dsmr/reading/phase_voltage_l3 232.0
dsmr/reading/phase_voltage_l2 233.0
dsmr/reading/phase_voltage_l1 232.0

dsmr/telegram/id 38860468
dsmr/reading/phase_voltage_l3 232.0
dsmr/reading/phase_voltage_l2 232.0
dsmr/reading/phase_voltage_l1 231.0
```
In de grafieken zitten wel decimalen.
[image]

dennissiemensma, 2020-05-07:
Kun je zien of je in de admin in `/admin/dsmr_mqtt/message/` die waardes ziet (kan zijn dat je flink moet refreshen als MQTT ze snel verwerkt) en dan aangeven of die daar ook al afgerond zijn?

Het is ook mogelijk dat het systeem van `232` altijd `232.0` maakt voor de MQTT-berichten, maar ik zou wel verwachten dat je af en toe een decimaal krijgt.

Mijn eigen test is trouwens met een nep-telegram, ik heb zelf geen v5 meter.

TopdRob, 2020-05-07:
Ik denk dat het misschien aan mijn meter ligt want in de reading is het ook al afgerond.
Ik heb een Landys+Gyr E350 DSMR 5.0.

[image]

DSMR-reader
WELKOM, ADMIN. WEBSITE BEKIJKEN / WACHTWOORD WIJZIGEN / AFMELDEN
Voorpagina › Datalogger › DSMR-uitlezing (alleen lezen) › 38861441 @ 2020-05-07 21:56:41+02:00
DSMR-uitlezing (alleen lezen) weergeven
GESCHIEDENIS

ID:
38.861.441
Timestamp:
7 mei 2020 21:56
Moment waarop de meting is gedaan, volgens de meter
Electricity delivered 1:
4.022,665
Meterstand van verbruikte elektriciteit (laagtarief) in kWh
Electricity returned 1:
0,000
Meterstand van teruggeleverde elektriciteit (laagtarief) in kWh
Electricity delivered 2:
4.974,791
Meterstand van verbruikte elektriciteit (piektarief) in kWh
Electricity returned 2:
0,000
Meterstand van teruggeleverde elektriciteit (piektarief) kWh
Electricity currently delivered:
0,260
Huidig elektriciteitsverbruik in kW
Electricity currently returned:
0,000
Huidige teruglevering elektriciteit in kW
Phase currently delivered l1:
0,257
Huidig elektriciteitsverbruik in fase L1 (in kW)
Phase currently delivered l2:
0,000
Huidig elektriciteitsverbruik in fase L2 (in kW)
Phase currently delivered l3:
0,002
Huidig elektriciteitsverbruik in fase L3 (in kW)
Extra device timestamp:
7 mei 2020 21:55
Tijdstip van meting door de externe (gas)meter
Extra device delivered:
3.647,823
Meterstand van de externe (gas)meter
Processed:
True
Geeft aan of deze meting al verwerkt is in het samenvoegen van statistieken
Phase currently returned l1:
0,000
Huidige elektriciteitteruglevering in fase L1 (in kW)
Phase currently returned l2:
0,000
Huidige elektriciteitteruglevering in fase L2 (in kW)
Phase currently returned l3:
0,000
Huidige elektriciteitteruglevering in fase L3 (in kW)
Phase voltage l1:
233,0
Huidig voltage in fase L1 (in V)
Phase voltage l2:
234,0
Huidig voltage in fase L2 (in V)
Phase voltage l3:
230,0
Huidig voltage in fase L3 (in V)
Phase power current l1:
1
Huidige stroomsterkte in fase L1 (in A)
Phase power current l2:
0
Huidige stroomsterkte in fase L2 (in A)
Phase power current l3:
0
Huidige stroomsterkte in fase L3 (in A)
Sluiten

dennissiemensma, 2020-05-07:
Je zou nog onderaan de pagina `/admin/dsmr_datalogger/meterstatistics/1/change/` het laatste telegram kunnen bekijken om te zien of de waardes daar wel een niet-0 decimaal hebben.

TopdRob, 2020-05-07:
Alle telegrammen hebben daar een 0 decimaal. Sorry voor het verdoen van je tijd.

dennissiemensma, 2020-05-07:
Bedankt voor je snelle terugkoppeling. In dat geval is dat ook goed voor mij om te weten.

Je zou toch nog even kunnen kijken of je niet ooit een decimale waarde hebt gehad:

```
sudo su - postgres
psql dsmrreader

select distinct(phase_voltage_l1)  from dsmr_consumption_electricityconsumption;
select distinct(phase_voltage_l2)  from dsmr_consumption_electricityconsumption;
select distinct(phase_voltage_l3)  from dsmr_consumption_electricityconsumption;
```

closed 2020-05-21 by dennissiemensma