dsmrreader/dsmr-reader #985: MQTT bericht voor overcapaciteit zonnepanelen

MORD0R, 2020-05-07:
Het zou handig zijn als je meer stroom opwekt dan je gebruikt. Dat hier een apart mqtt bericht van is.

Dit kan handig zijn als de saldering stopt in Nederland. Als je teveel opwerkt zou je de wasmachine bijvoorbeeld kunnen laten aangaan.

dennissiemensma, 2020-05-07:
Bedankt voor je verzoek. Ik denk dat je dit nu ook al zelf kunt detecteren met de huidige MQTT-waardes.

Een telegram heeft de velden `electricity_currently_delivered` en `electricity_currently_returned`. Wanneer de eerste 0 is en de tweede groter dan nul, ben je aan het terugleveren en kun je aan de hand daarvan een melding of iets anders triggeren.

closed 2020-05-07 by dennissiemensma

MORD0R, 2020-05-11:
Dankje ik ga het proberen sry voor de late reactie.