dsmrreader/dsmr-reader #988: Updaten is niet mogelijk

egrutter, 2020-05-10:
Hallo,
 bij het updaten, ./deploy.sh (in de virtualenv.) krijg ik de volgende log te zien:
"
 --- Pulling remote repository for new commits...
remote: Enumerating objects: 161, done.
remote: Counting objects: 100% (161/161), done.
remote: Compressing objects: 100% (85/85), done.
remote: Total 161 (delta 89), reused 119 (delta 63), pack-reused 0
Receiving objects: 100% (161/161), 112.05 KiB | 726.00 KiB/s, done.
Resolving deltas: 100% (89/89), completed with 8 local objects.
error: inflate: data stream error (incorrect data check)
fatal: packed object 640083f85bb78e246eaa200a7867eb43911260f5 (stored in .git/objects/pack/pack-0d9c9f24d3a058a13189d7e174e818e6addcd568.pack) is corrupt
error: https://github.com/dennissiemensma/dsmr-reader.git did not send all necessary objects
"

Hierbij blijft hij op versie 3.8 hangen. 
(heb de vorige keer niet goed opgelet en ./deploy.sh uitgevoerd en er vanuit gegaan dat het geupdated was.)

Komt dit vaker voor?  Weet iemand of dit aan mijn kant ligt of dat er (onderweg) bij git iets niet goed zit?

Groet, Erwin

Ik draai op een raspberry pi 3+, Raspbian Buster Lite.

dennissiemensma, 2020-05-11:
Bedankt voor je vraag. Ik heb deze fout zelf nog niet eerder gehad, maar het lijkt in je lokale kopie van dit project te zitten.

Afhankelijk van hoe technisch je bent, kun je proberen om dit zelf op te lossen, bijvoorbeeld met: https://gist.github.com/sebastianbachmann/86ebb6dd5a0270e23725

----

Als je er helemaal niet thuis in bent, kun je de lokale kopie van de code (alleen de code en niet de database!) verwijderen en opnieuw uitchecken.

```
sudo supervisorctl stop all
sudo su - dsmr

mv dsmrreader/settings.py ~/
mv backups/ ~/

cd ..
rm -rf dsmr-reader
git clone https://github.com/dennissiemensma/dsmr-reader.git dsmr-reader

mv ~/settings.py dsmr-reader/dsmrreader/
sudo supervisorctl start all
```

Na afloop staan je backups nog in home/dsmr/backups/ Die kun je nog proberen om terug te plaatsen in de map of je wacht even een nachtje totdat je een nieuwe backup hebt. Daarna kun je je oude ook weggooien als dat makkelijker is.

egrutter, 2020-05-11:
Update: het zat vast in mijn lokale kopie, na bovenstaande te hebben uitgevoerd, functioneert hij weer naar behoren, incl de nieuwste update.

Ben (nog) niet zo thuis in git e.d. en heb bovenstaande gevolgd en dat werkt perfect. 
Bedankt voor de moeite en het antwoord!

closed 2020-05-11 by egrutter