dsmrreader/dsmr-reader #989: Totale kosten maand

poudenes, 2020-05-10:
Zou het mogelijk zijn om de totale kosten van de desbetreffende maand via MQTT uitgezonden kunnen worden, en eventueel de totale kosten met start datum dat in gevoerd is in

Voorpagina › Verbruik › Tarieven energieleverancier

Dan heb je 2 MQTT topics erbij om over gehele contract periode totalen bij te houden.
En eentje die geldt alleen voor de lopende maand.

dennissiemensma, 2020-05-11:
Bedankt voor je verzoek. Ik zal het overwegen voor een toekomstige release.

giejay, 2020-06-01:
Dit lijkt me ook zeer interessant, naast simpelweg het totale verbruik per maand

lubbertkramer, 2020-10-27:
Volgend, lijkt mij een mooie toevoeging!

dataegg, 2020-12-26:
Ik zou deze toevoeging ook op prijs stellen. Op die manier kan ik ook het energie verbruik voor de huidige maand in home assistant weergeven en ik hoop daarmee te bereiken dat gezinsleden meer bewust met energie omgaan.

skank01, 2021-02-11:
per maand en per jaar ook eigenlijk

skank01, 2021-02-13:
Eigenlijk bij nader inzien, alles van "per maand" he, zowel het verbruik laag/hoog tarief, verbruik gas, maar ook de kosten
Dit van de huidige maand, en de vorige maanden ... (per jaar?)

dennissiemensma, 2021-02-13:
Bedankt voor je aanvulling. Ik denk eerlijk gezegd dat MQTT **niet** bedoeld is voor het ophalen of exporteren van historische gegevens zoals vorige maanden.
Het mechanisme doet updates van iets wat verandert, de huidige situatie. Dat kan in de context zijn van de huidige minuut, huidige dag, huidige maand of huidig jaar, maar niet de vorige maand of een paar maanden terug, want die veranderen niet meer.

Ik ben sowieso van plan om alle losse MQTT-wensen te herschrijven naar 1 algemene wens, met de mogelijkheden die DSMR-reader nu ook toont in het dashboard, plus de som van het huidige jaar. Ik vermoed dat de meesten daar mee geholpen zijn en die gegevens lenen zich uitstekend voor MQTT omdat ze dagelijks veranderen (na het genereren van de statistieken van de voorgaande dag).

skank01, 2021-02-13:
Mmm je hebt een punt.
Maar hoe krijg ik dan in ha een graph van het verbruik per maand?
Met de utility meter in ha, krijg je een vertekend beeld wanneer je bv
utility_meter:
monthly_energy_offpeak:
source: sensor.energy_consumption_tarif_1
cycle: monthly

doet

dennissiemensma, 2021-02-13:
Ik denk dat het goed is om te weten dat DSMR-reader er niet voor kan zorgen dat een andere tool een bepaalde visualisatie kan maken van de gegevens in DSMR-reader. 
Dit project biedt MQTT voor het pushen van data-updates en een REST API voor het pullen van alle opgeslagen data. Daarmee zou het gros van alle mogelijke wensen te realiseren moeten zijn. Helemaal als daar straks ook nog een dagelijkse MQTT update bij komt voor de tussenstand van de huidige maand/jaar.

Een mogelijke oplossing zou zijn om de dagtotalen via MQTT op te slaan en dat zelf te aggregeren/plotten in HA. Of om de REST API te gebruiken en te zoeken naar dagstatistieken in een maand en die op te tellen. In beide gevallen zou je dan de gegevens hebben voor de grafieken (lijkt me, zonder te weten hoe HA intern werkt).

dennissiemensma, 2021-02-13:
Dit issue is nu opgenomen in #1291.

closed 2021-02-13 by dennissiemensma

dennissiemensma, 2021-03-07:
Dit gaat mee bij de aanstaande v4.13-release. Naast de gegevens via MQTT worden ze ook zichtbaar in het dashboard. Zie #1291 voor meer informatie en screenshots.

giejay, 2021-03-07:
Dankje Dennis!