dsmrreader/dsmr-reader #990: Feature request - Alerts genereren obv energie verbruik

ph0real, 2020-05-11:
Hoi Dennis,

Al een flinke tijd gelukkige gebruiker van DSMR; waarvoor dank.
Ik vroeg me vanochtend af, hoe interessant het zou zijn als er een feature bestond om een gebruiker te waarschuwen bij een bepaald energie verbruik op een gegeven moment. Een voorbeeld; ik weet dat ik na 23:00 xx energie verbruik en niet veel meer. Maar ik heb een apparaat aan laten staan en het verbruik ligt dus hoger. Met deze feature zou ik dmv triggers en alerts op de hoogte gesteld kunnen worden dat ik iets vergeten ben uit te zetten in het huis. 

Ik weet niet of je deze kant op wil met de datalogger, maar dacht, vraag het toch even.

Gr. Dennis

dennissiemensma, 2020-05-11:
Bedankt voor je verzoek. Ik denk dat inderdaad wat te specifiek is voor dit project zelf.

Ik denk dat er al voldoende tools zijn binnen het project. Je kunt via MQTT een push krijgen van het huidige verbruik en daar een alert op maken. Of je kunt de API gebruiken en op vaste intervallen opvragen wat het verbruik is en dan een alert maken.

ph0real, 2020-05-11:
Ah ik wist niet dat t kon met deze alternatieven. Ik zal er es induiken!

closed 2020-05-21 by dennissiemensma