dsmrreader/dsmr-reader #991: Extern toegang ios snelkoppeling werkt niet *goed*

aroe0546, 2020-05-11:
Wanneer ik via mijn standaart ios browser naar mijn ip ga om dsmr te openen gaat alles prima wordt netjes gevraagd om inlog naam en wachtwoord. Wanneer ik een snelkoppeling maak naar het start scherm blijft het beeld zwart na het inloggen. nu ben ik wat gaan testen kwam ik er achter wanneer ik een snelkoppeling maak voor dat ik mijn inlog naam en wachtwoord heb ingevoerd  dus met de pop up er voor dan werkt het wel maar moet ik dus telkens mijn wachtwoord etc. invoeren erg vervelend!   Ook werkt de favo icon niet en heb ik dus een leeg icoontje op mijn start scherm.

dennissiemensma, 2020-05-11:
Bedankt voor je melding. Ik heb zelf geen iOS toestellen, dus dat is wat lastig voor mij om op te lossen qua icon. Ik heb echter in #249 zoiets eerder ingesteld, dus ik kan altijd kijken of de specs van Apple zijn veranderd. 

Wat betreft bugs met een zwart scherm en wachtwoorden die wel of niet bewaard worden, dat is helaas iets van het OS zelf en kan ik waarschijnlijk weinig in betekenen.

dennissiemensma, 2020-05-28:
Ik zie geen wijzigingen in de docs van Apple, maar ik kan het helaas ook niet voor je reproduceren hier. Hoewel het voor de meeste devices zou moeten werken, kan ik ze niet allemaal testen. Voor nu laat ik het voor wat het is.

closed 2020-05-28 by dennissiemensma