dsmrreader/dsmr-reader #992: Update van 2.1 naar 2.15 gaat fout

stefanroelofs, 2020-05-13:
Ik had versie 2.1.
Update gestart met ./deploy.sh
Gaat mis op deze regel in post-deploy.sh:
pip3 install -r dsmrreader/provisioning/requirements/base.txt --upgrade

foutmelding:
```
(dsmrreader) dsmr@dsmr-reader:~/dsmr-reader $ pip3 install -r dsmrreader/provisioning/requirements/base.txt --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r dsmrreader/provisioning/requirements/base.txt (line 20))
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x75d598d0>, 'Connection to qpypi3.qpython.org timed out. (connect timeout=15)')': /repository/1745924/PyCRC-1.21.tar.gz
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))': /repository/1745924/PyCRC-1.21.tar.gz
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='qpypi3.qpython.org', port=443): Read timed out. (read timeout=15)",)': /repository/1745924/PyCRC-1.21.tar.gz
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='qpypi3.qpython.org', port=443): Read timed out. (read timeout=15)",)': /repository/1745924/PyCRC-1.21.tar.gz
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))': /repository/1745924/PyCRC-1.21.tar.gz
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='qpypi3.qpython.org', port=443): Max retries exceeded with url: /repository/1745924/PyCRC-1.21.tar.gz (Caused by ProtocolError('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer')))

You are using pip version 19.0.2, however version 20.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
PyCRC-1.21.tar.gz lijkt niet gevonden te kunnen worden?

Zie ook issue #850 

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v2.1
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3b
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v4

dennissiemensma, 2020-05-13:
Bedankt voor je vraag. Ik denk dat je het beste verder kunt upgraden naar v3.x, maar dan moet je wel mogelijk je Python-versie bijwerken: https://dsmr-reader.readthedocs.io/en/latest/faq/v3_upgrade.html

Die PyCRC was destijds van de ene op de andere dag niet meer beschikbaar en de mirror nu dus blijkbaar ook niet meer.

stefanroelofs, 2020-05-14:
Bedankt voor je reactie. 
De update naar v3 ging zonder problemen! 👍

closed 2020-05-14 by stefanroelofs