dsmrreader/dsmr-reader #993: Back-up terugzetten lukt niet

Gercovo, 2020-05-14:
Ik heb een clean-install op mijn Raspberry Pi3 uitgevoerd en ik wil graag een backup terugzetten, maar ik krijg de melding "invalid command \N" en uiteindelijk "out of memory". Ik had altijd een back-up lopen via Dropbox. Het bestand is ingepakt wel 353MB groot.

Hierbij de link op deze te downloaden.

https://dl.dropbox.com/s/goekp7bsv9qj92i/dsmrreader-postgresql-backup-Monday.sql.gz?dl=0

Eggenhuizen, 2020-05-14:
wellicht overbodig maar.. 

@Gercovo ben je je ervan bewust dat je jouw persoonlijke energie verbruiksdata nu publiekelijk beschikbaar stelt?

Gercovo, 2020-05-14:
Ik ben mij er van bewust, maar indien het niet nodig is om deze te delen dan liever niet natuurlijk. Dus ik heb het maar even (tijdelijk) verwijderd.

dennissiemensma, 2020-05-14:
Bedankt voor je vraag. Ik denk dat je backup simpelweg te groot is geworden voor je Pi. Ik gok dat die 350 MB uitgepakt vele GB's zijn en dat die daar tegenaan loopt. 
Ik kan je sowieso aanraden om straks retentie in te stellen zodat de dataset wat kleiner is, helemaal omdat je vermoedelijk DSMR-v5 meter hebt.

Ben je in de gelegenheid om de backup zelf ergens anders uit te pakken?

Gercovo, 2020-05-14:
Uitgepakt is het circa 2,5GB. Ik kan de back-up zeker zelf even uitpakken. Dit zal ik nu even doen.

dennissiemensma, 2020-05-14:
Ooh dat valt me eerlijk gezegd wel mee qua grootte. In dat geval kun je het eens proberen om de import te doen met het uitgepakte bestand, met de onderste variant: https://dsmr-reader.readthedocs.io/en/latest/installation/restore.html

Gercovo, 2020-05-14:
Helaas komt hij dan op een gegeven moment weer met "out of memory" .

dennissiemensma, 2020-05-14:
Bedoel je bij het uitpakken of bij het importeren?

Gercovo, 2020-05-14:
Excuus, het importeren.

dennissiemensma, 2020-05-14:
Ik ben daar helaas zelf nog niet tegenaan gelopen en zie op Google een hoop verschillende oorzaken en oplossingen daarover. Je kunt daar nog wat van proberen, maar ik heb geen gegarandeerde fix voor je.

Het enige wat ik zelf kan bedenken is dat je de dump ergens anders probeert in te lezen waar mee geheugen beschikbaar is en DSMR-reader er dan retentie op laat toepassen. Ik kan het eventueel ook voor je proberen, maar bij voorkeur doe je het natuurlijk zelf.

dennissiemensma, 2020-05-21:
@Gercovo ben je hier nog verder mee gekomen?

closed 2020-05-28 by dennissiemensma

Gercovo, 2020-06-09:
Excuus voor de late reactie. Helaas is het mij nog niet gelukt. De oplossing die jij aandraagt is daar een uitleg van te vinden?

dennissiemensma, 2020-06-09:
Je zult daarvoor eerst een PC of laptop of server moeten hebben waar je de database op kan importeren. En anders kan ik het eventueel hier op mn PC voor je proberen.

Gercovo, 2020-06-10:
Dat idee had ik al, is dat https://www.postgresql.org/ downloaden en installeren?

Gister dacht ik laat ik het nog eerst een keer proberen, nu kreeg ik niet de melding "out of memory", maar aan het einde gewoon "killed". Wanneer ik nu naar het dashboard ga en dan naar archief ga staat er "Geen gegevens gevonden".