dsmrreader/dsmr-reader #994: Nieuwe meter ontvangen, alle standen in de min.

steffjenl, 2020-05-15:
Alle standen op mijn dashboard staan ver in de min na dat mijn elektra en gas meter zijn vervangen. Er staat nergens een handleiding wat je moet doen wanneer de meter is vervangen. Graag zou ik mijn historie willen behouden en vanaf vandaag verder gaan met de nieuwe standen.

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.9.1
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3+
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

dennissiemensma, 2020-05-15:
Bedankt voor je vraag. Als het goed is komt het maar voor 1 dag voor, daarna gaat het goed. Voor de ene foute dag kun je het handmatig in de admin aanpassen: https://github.com/dennissiemensma/dsmr-reader/issues/507#issuecomment-407127424

steffjenl, 2020-05-15:
Bedankt Dennis! misschien idee om dit in de FAQ te plaatsen? Dat iemand verhuisd kan nog wel eens voorkomen. Nieuwe meter misschien niet?

dennissiemensma, 2020-05-15:
Ik zal het opnemen in de FAQ.

dennissiemensma, 2020-05-21:
Toegevoegd aan de FAQ bij volgende release.

closed 2020-05-21 by dennissiemensma