dsmrreader/dsmr-reader #997: Meterstanden van een bepaalde dag.

Edwink18, 2020-05-19:
Hoe kan ik zien wat de meterstanden op een bepaalde dag geweest zijn.  Ik kan wel het gebruik terug vinden in het archive maar de exacte meterstanden kon niet terug vinden. 

Aanleiding was de melding van de energiemaatschappij dat ze de meterstanden hadden uitgelezen, wilde dit even controleren.

TomWis97, 2020-05-19:
Dat kan je op zich prima uit de database halen. Start een sessie met je database (`postgres` user, `psql` commando.) En doe zoiets:
1. Laat alle databases zien: `\l` Kijk vervolgens welke je gebruikt voor dsmr-reader.
1. Open de database met `\c`. Bijvoorbeeld `\c dsmrreader`.
1. Voer de select uit voor de gegevens van de dag, deze query laat de eerste 5 resultaten zien van 16-5-2020: `SELECT * FROM dsmr_consumption_electricityconsumption WHERE read_at::date = date '2020-05-16' limit 5;`

dennissiemensma, 2020-05-21:
Bedankt voor je vraag. Je kunt het inderdaad uitlezen zoals Tom aangeeft. 

Een alternatief is om via `/admin/dsmr_datalogger/dsmrreading/?p=1` deze data te bekijken, maar daar zit (nog) geen zoekfunctie in.

dennissiemensma, 2020-05-21:
Voor de volgende release heb ik het mogelijk gemaakt om binnen de admin te kunnen zoeken naar een bepaalde dag (range). Dan hoeft dit niet meer perse handmatig via de database.

closed 2020-05-21 by dennissiemensma

dennissiemensma, 2020-05-21:
Ter verheldering een screenshot. Aan de rechterkant zit straks een zoeker hiervoor. Daarmee kun je vrij gemakkelijk de gegevens voor 1 dag of moment filteren.

[image]

Edwink18, 2020-05-21:
Prima idee.  ik heb meerdere DSMR image's draaien echter op 1 van de image's deze  loopt al lang en heeft een logging van elke sec.  Krijg ik een gataway timeout nu als ik naar dit scherm wil.  Default staat het filter nu op any.  Dat geeft denk ik teveel data uit de database dat dit nu een timeout veroorzaakt.  Misschien handiger om de query pas na de filter instelling te doen.

dennissiemensma, 2020-05-21:
Wellicht wil je nog het automatisch opschonen van oude metingen aanzetten (retentie). Als de dataset te groot wordt, dan loop je op den duur tegen dit soort issues aan.