dsmrreader/dsmr-reader #998: Teveel onverwerkte telegrammen

R0yk3, 2020-05-22:
**Wat gebruik je?**
* DSMR-reader versie: v. 3.9.1
* Hardware: Debian 10 laptop 
* Native of Docker: Native install Postgres 11.8
* DSMR-protocol slimme meter: v5

[https://prnt.sc/slstwc](https://prnt.sc/slstwc)

Zoals je ziet loopt het al een tijdje achter (ik kijk er niet vaak naar) 
Logging aangezet en deze melding blijft zich herhalen.

> The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 369, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 45, in handle
    self._write_pid_file()
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 135, in _write_pid_file
    with open(self._pid_file, 'w') as file_handle:
PermissionError: [Errno 13] Permission denied: '/var/tmp/dsmrreader--dsmr_backend.pid'

Deze melding is overigen na de reboot niet meer terug gekomen. er zijn nu geen nieuwe logmeldingen.
Alleen supervisord.log geeft meldingen.

> 2020-05-22 14:11:19,640 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-05-22 14:11:20,644 INFO spawned: 'dsmr_datalogger' with pid 25282
2020-05-22 14:11:22,154 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-05-22 15:11:22,447 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-05-22 15:11:23,451 INFO spawned: 'dsmr_datalogger' with pid 26330
2020-05-22 15:11:24,965 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-05-22 16:11:25,259 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-05-22 16:11:26,295 INFO spawned: 'dsmr_datalogger' with pid 27577
2020-05-22 16:11:27,842 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

Ieder uur 3 regels maar dat zijn mijn inziens geen foutmeldingen

Wat nu?? 
hoe krijg ik hem weer up to date zonder alles weg te gooien en opnieuw te starten.

Alvast bedankt

dennissiemensma, 2020-05-23:
Bedankt voor je vraag. He bje ook de debug-logging aangezet, zoals in https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging?

Als het goed is moet je dan in de log van de backend, in de supervisor log-map, kunnen zien wat er gebeurt en wat eventuele fouten zijn.

R0yk3, 2020-05-23:
ja heb ik gedaan,
de output daarvan staat in de eerste post.
Verder worden de log bestanden niet meer bijgewerkt dus de laatste melding is van gisteren. na een reboot komen er geen nieuwe meldingen met uitzondering van meldingen in dit bestand:

/var/log/supervisor/dsmr_datalogger-stderr---supervisor-2QGXqL.log

> ERROR:commands:dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exceptio                                                           n raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/se                                                           rialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogg                                                           er.py", line 21, in run
    dsmr_datalogger.services.datalogger.read_and_process_telegram()
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 69,                                                            in read_and_process_telegram
    read_serial_port(**connection_parameters)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_clien                                                           t.py", line 51, in read_serial_port
    timeout=20,  # Max time to wait for data.
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/se                                                           rialutil.py", line 240, in __init__
    self.open()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/se                                                           rialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._p                                                           ort, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [                                                           Errno 2] No such file or directory: '/dev/ttyUSB0'

[2020-05-23 14:00:29,676] INFO     dsmr_datalogger.management.commands.dsmr_data                                                           logger: Exiting on next run...
INFO:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on ne                                                           xt run...
[2020-05-23 14:00:29,677] DEBUG    dsmr_datalogger.management.commands.dsmr_data                                                           logger: Sleeping 0.5s
DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5                                                           s
[2020-05-23 14:00:29,677] DEBUG    dsmr_datalogger.management.commands.dsmr_data                                                           logger: Exited
DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exited
[2020-05-23 14:00:47,335] DEBUG    dsmr_datalogger.management.commands.dsmr_data                                                           logger: Starting infinite command loop...

Hieruit concludeer ik dat er geen nieuwe data wordt uitgelezen vanaf mijn P1 connectie.
Maar daarmee wordt de backlog niet kleiner.

gr
Roy

dennissiemensma, 2020-05-24:
Bedankt voor je aanvulling. Als het achtergrond proces goed draait, dan zul je DEBUG-statements moeten zien in `/var/log/supervisor/dsmr_backend.log`. 
Zie je die niet, dan staat ofwel de logging niet op DEBUG ofwel het proces kan uberhaupt niet starten. In de eerste log zie je iets met `/var/tmp/`, die map moet sowieso bestaan en schrijfbaar zijn. Dat kun je ook nog checken, hoewel je aangeeft dat die niet meer vaker voor komt na de reboot.

Heb je trouwens wel voldoende schijfruimte? Dat zorgt soms ook voor hele vage issues: `df -h`

dennissiemensma, 2020-05-24:
En de andere fout gaat over de datalogger, het lijkt erop dat de poort `/dev/ttyUSB0` niet bereikbaar is (of hernoemd is?). Je kunt kijken of je die nog wel met `cu` [handmatig kunt uitlezen](https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional).

R0yk3, 2020-05-24:
De datalogger staat aan, tenminste ik heb de stappen gevolgd zoals aangegeven.
Ruimte is er voldoende 

> roy@domoticz:/$ df -h
Filesystem                Size  Used Avail Use% Mounted on
udev                      3.9G     0  3.9G   0% /dev
tmpfs                     790M   37M  753M   5% /run
/dev/sda1                 450G   22G  405G   6% /

Nu wordt er ook weer in het log geschreven, tenminste ik zie aan de datum van het bestand dat het gebruikt wordt, in de log wordt geen datum weggeschreven.
Onderstaand is de melding:
>Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 369, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 45, in handle
    self._write_pid_file()
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 135, in _write_pid_file
    with open(self._pid_file, 'w') as file_handle:
PermissionError: [Errno 13] Permission denied: '/var/tmp/dsmrreader--dsmr_backend.pid'

De var directory bestaat, daar staan ook diverse PID files in.
Eiegnaar van dsmrreader--dsmr_backend.pid is gebruiker 1001 en niet dsmr.

Zou het dat zijn??

de tty kijk ik later naar..

dennissiemensma, 2020-05-24:
Je kunt in dat geval die eigenaar wijzigen naar `dsmr`. Grote kans dat dat het direct oplost.

R0yk3, 2020-05-24:
Dit Ziet er gezond uit:

> [2020-05-24 14:46:57,733] DEBUG    dsmr_backend.management.commands.dsmr_backend: Starting infinite command loop...
[2020-05-24 14:46:57,773] DEBUG    SP: 7 backend service(s) ready to run
[2020-05-24 14:46:57,773] DEBUG    SP: Running "Backup per email" (dsmr_backup.services.email.run)
[2020-05-24 14:46:57,873] DEBUG    Email backup: Interval not set, skipping backup for a day
[2020-05-24 14:46:57,940] DEBUG    SP: Rescheduled "Backup per email" to 2020-05-25 14:46:57.873792+02:00 (ETA 23:59:59.933044)
[2020-05-24 14:46:57,940] DEBUG    SP: Running "Create daily backup" (dsmr_backup.services.backup.run)
[2020-05-24 14:46:58,016] INFO      - Creating non-existing backup folder: /home/dsmr/dsmr-reader/backups/archive/2020/05
[2020-05-24 14:46:58,016] INFO      - Creating new partial backup: /home/dsmr/dsmr-reader/backups/archive/2020/05/dsmrreader-postgresql-partial-backup-2020-05-24.sql
[2020-05-24 14:46:58,758] DEBUG     - Backup exit code: 0
[2020-05-24 14:46:58,761] INFO      - Created and compressed statistics backup: /home/dsmr/dsmr-reader/backups/archive/2020/05/dsmrreader-postgresql-partial-backup-2020-05-24.sql.gz
[2020-05-24 14:46:58,763] INFO      - Creating new full backup: /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Sunday.sql
[2020-05-24 14:48:53,417] DEBUG     - Backup exit code: 0

R0yk3, 2020-05-24:
Gaat hij nu automatisch de oude telegrammen verwerken??

dennissiemensma, 2020-05-24:
Je kunt proberen om je poort te klonen met `socat`, zodat zowel DSMR-reader als Domoticz allebei een eigen poort hebben en elkaar niet in de weg zitten: https://github.com/dennissiemensma/dsmr-reader/issues/377#issuecomment-442237344

R0yk3, 2020-05-24:
Dat is even een paar stappen te ver ;)
met het CU commando krijg ik wel de melding connected, maar daar blijft het dan wel bij.

> (dsmrreader) dsmr@domoticz:~/dsmr-reader$ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.

Zo staat hij nu enkele minuten.. 
Ik snuffel even verder.
Domoticz heb ik uitgezet. het zou moeten werken want dat doet het al heel lang naast elkaar...

Hints en tis zijn welkom.

dennissiemensma, 2020-05-24:
Het kan zijn dat dat in nieuwere versies van DSMR-reader niet meer samen werkt, omdat de datalogger is aangepast rond DSMR-reader v3.3. 
Niet om die reden trouwens, maar voor de performance van het uitlezen, die relatief zwaar was.

Als je DSMR-reader en Domoticz uit hebt staan en het dan nog steeds niet met `cu` werkt, weet ik ook niet even 1-2-3 wat het kan zijn. Wellicht dubbelchecken of de user waarmee je dat doet wel toegang heeft tot de `dialout` groep?

R0yk3, 2020-05-27:
Dag, ik heb nu apart een logger geinstalleerd. deze werkt. ik heb ook cu geinstalleerd en ik kan allle telegrammen zien.
Alleen ik zie nog niets verschijnen op mijn reader.
Api staat aan
ingevulde URL in dsmr_datalogger_api_client.py 
('http://http://192.168.1.253:8090//api/v1/datalogger/dsmrreading', 'Authenticatiesleutel'),  

URL klopt want als ik 
curl 'http://192.168.1.253:8090/api/v2/datalogger/dsmrreading?timestamp__gte=2017-02-01%2000:00:00&timestamp__lte=2017-03-01%2000:00:00' \
    -H 'X-AUTHKEY: Authenticatiesleutel' | python -m json.tool
invoer op mijn logger dan krijg ik respons. data is nul want in 2017 had ik hem nog niet draaien.

Maar per saldo worden de telegrammen niet verwerkt???

dennissiemensma, 2020-05-27:
Wat geeft de Status-pagina aan? Staan daar onverwerkte telegrammen of helemaal niets?

En wat voor melding staat er in de logs van de datalogger? Afhankelijk van hoe je datalogger in Supervisor heet (standaard `dsmr_remote_datalogger`) zou deze log moeten bestaan. 

```
/var/log/supervisor/dsmr_remote_datalogger.log
```

Staat daar iets in?

R0yk3, 2020-05-27:
dsmr_remote_datalogger.log Dit bestand bestaat niet.
meldingen pagina geeft alleen aan dat 

> "Het is een tijdje geleden sinds het laatste telegram uitgelezen is."

Moet ik in web ui nog iets meer instellen dan alleen bij de API config, API calls toestaan?

R0yk3, 2020-05-27:
Ik denk in ieder geval al 1 foutje te hebben gevonden. De URL in [](https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py)
is api/v1
Mijn test was met api/v2
Die laatste geeft respons, v1 geeft geen respons.
Ik zie niet direct data maar ik laat het even lopen.
Ik heb ook nog een volledig 'schone' reader opgetuigd dan weet ik zeker dat het niet aan de installatie ligt.

dennissiemensma, 2020-05-27:
Je kunt nog overwegen om Postman te installeren en gebruiken. Het is een vrij makkelijke tool voor API's ontwikkelen en testen. 
Verder staat er een complete Postman-collectie + variabelen klaar in je lokale DSMR-reader installatie op: `/docs/api#section/Postman-examples`. Na import in Postman hoef je als het goed is alleen je hostname en API-key als variabele rechtsbovenin in te stellen.

Als de v2 API wel iets teruggeeft zou de v1 API ook moeten werken.

Heb je nou zowel DSMR-reader als de remote datalogger op je laptop draaien? Of heb je DSMR-reader op je laptop en de remote-datalogger op bijvoorbeeld een Pi aan je meter?

Als je de remote datalogger in Supervisor hebt draaien, verwacht ik dat deze file er wel moet zijn: `/var/log/supervisor/dsmr_remote_datalogger.log`. Maar dus op het apparaat van de remote datalogger. We hebben het wel over dezelfde datalogger toch? https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html

R0yk3, 2020-05-27:
reader staat op mijn laptop. datalogger op een pi.
Ik was op mijn reader aan het zoeken naar het log bestand. maar dat staat op de logger.
dit staat erin 

> File "/home/dsmr/dsmr_datalogger_api_client.py", line 21
    :SERIAL_SETTINGS = dict(
    ^
En ja ik heb de versie gebruikt waar je naar verwijst

dennissiemensma, 2020-05-27:
Die foutmelding geeft aan dat je een typo hebt gemaakt in `/home/dsmr/dsmr_datalogger_api_client.py`. 

Die dubbele punt vooraan de regel hoort daar niet en mag je weghalen in het bestand. Probeer het daarna nogmaals, met `sudo supervisorctl restart all` op de Pi.

R0yk3, 2020-05-27:
Dan krijg ik de onderstaande melding

> dsmr_remote_datalogger: ERROR (spawn error)

R0yk3, 2020-05-27:
YES!!!!

R0yk3, 2020-05-27:
V2 werkt niet om telegrammen te verzenden. ik heb er weer v1 van gemaakt.
Tussen de Nl en de EN versie zit wel een verschil (zo op het eerste gezicht)
maar in ieder geval bedankt. Ik heb een pul bier gedoneerd. Zodra het terras open gaat, proost... :D

dennissiemensma, 2020-05-28:
Fijn om te horen dat het gelukt is!

Voor de duidelijkheid: De v1 en v2 API zijn dus verschillend en dat zou ook in de API-docs moeten staan. Je kunt naar de v1 API de complete telegrammen versturen en naar de v2 API alleen de losse velden. 
De remote datalogger is gemaakt voor v1, die URL staat ook hardcoded in het script, dus je kunt daar inderdaad niet zomaar v2 van maken.

Wat betreft verschillen tussen NL en EN, als je iets significants ziet wat niet klopt dan hoor ik het graag zodat ik het kan aanpassen. :]

closed 2020-05-28 by dennissiemensma

R0yk3, 2020-05-28:
Dag Dennis,
Ik zien of er iets niet klopt ;) ik ben een scriptkiddie of meer een
scriptoldie.
Maar ik zal eens mijn best doen.

Gr
Roy