dsmrreader/dsmr-reader #1000: Retentie standaard inschakelen

dennissiemensma, 2020-05-25:
Dit omdat er afgelopen maanden regelmatig issues zijn gemeld door o.a. grote datasets, wat herstel en debugging onnodig moeilijk maakt. 
Voor bestaande installaties kan die denk ik uit blijven, wellicht even een melding weergeven om men erop te attenderen.

dennissiemensma, 2020-07-29:
Ingeschakeld voor nieuwe installaties vanaf v4.1

dennissiemensma, 2020-07-29:
Ook nog even een eenmalige dashboard-melding toegevoegd voor als iemand dit uit heeft staan. Ter informatie.

dennissiemensma, 2021-03-24:
# eCharts 5.x + restyling grafieken
- donnie-dominguez

cmderks93, 2021-03-25:
Top, zal het vanavond installeren :-)

cmderks93, 2021-03-26:
Dennis,
Kan ik jou ook een privé bericht sturen?

cmderks93, 2021-03-26:
Mijn eerste indruk is helaas niet zo goed.
Het is wat mij betreft niet overzichtelijker. Wellicht dat een grafiek met een nullijn in het midden wat overzichtelijker is. Zodat het verbruik boven de nul komt, en het terugleveren onder de nul. Op deze manier komen de twee ook niet door elkaar te lopen.
[image]

dennissiemensma, 2021-03-26:
>  Kan ik jou ook een privé bericht sturen?

Zeker, mag naar github at dennissiemensma punt nl.

> Het is wat mij betreft niet overzichtelijker. Wellicht dat een grafiek met een nullijn in het midden wat overzichtelijker is. Zodat het verbruik boven de nul komt, en het terugleveren onder de nul. Op deze manier komen de twee ook niet door elkaar te lopen.

Mooi, want dat was ook de hoofdreden om te vragen of je het wilde proberen. Ik was namelijk al bang dat ze in jouw situatie elkaar overlappen. 
Ik denk dat je suggestie een goed alternatief is. Ik zal zo eens kijken of ik wat in elkaar kan draaien. Hier staan een shitload aan mogelijkheden: https://echarts.apache.org/examples/en/index.html

Wellicht dat iets in de richting van deze twee een oplossing biedt, waarbij ik de tweede het meest clean vindt (qua X-as):
- https://echarts.apache.org/examples/en/editor.html?c=grid-multiple
- https://echarts.apache.org/examples/en/editor.html?c=bar-animation-delay

cmderks93, 2021-03-26:
Mee eens, de tweede lijkt het mooiste.

dennissiemensma, 2021-03-26:
Ik heb de beta-versie bijgewerkt. Het is nog steeds de rainfall-variant, maar nu met twee losse Y-assen, zodat ze nooit door elkaar heen lopen. Verder nog wat kleine fixes.

Als je nog **niet** teruggewisseld was naar de v4.14-versie, kun je gewoon dit doen:
```
sudo su - dsmr
./deploy.sh
```

dennissiemensma, 2021-03-27:
Ik zie dat deze nieuwe variant al wat beter is, maar het voelt (voor mij) nog wat te ongerelateerd tussen beide assen. Ik wil nog wel een derde variant maken met die nul-as in het midden.

cmderks93, 2021-03-27:
Dennis,

Dat kan ik toejuichen. Ik vind de huidige idd nog niet DE oplossing.
Vandaag is ook een mooi dagje, met veel variabelen ivm de regen en zonneschijn.

dennissiemensma, 2021-03-27:
Ik heb weer een nieuwe beta-versie gepushed.

cmderks93, 2021-03-27:
En ik heb een deploy.sh uitgevoerd:-)
Stukken beter wat mij betreft :-)

donnie-dominguez, 2021-03-27:
[image]
[image]
Ik moet nog even wennen hoor, ik ga het morgen ook in de live omgeving proberen. Het lost in ieder geval het probleempje met de kleuren van het terugleveren op bij mij.

dennissiemensma, 2021-03-27:
@donnie-dominguez  bedankt voor het proberen. Ik kan me voorstellen dat het even wennen is, omdat je dan ook meteen alle wijzigingen in 1x ziet ipv in kleine stapjes.

Ik ben er zelf nog niet helemaal over uit qua "decals" (die driehoekjes/stippen/vierkantjes in de grafieken), dus ik heb die voor nu net weer uitgeschakeld in beta # 7. Je kunt bijwerken met `./deploy.sh`.

dennissiemensma, 2021-03-28:
Ik heb inmiddels de huidige opzet gemerged naar de develop branch voor de volgende release. Wellicht kom ik later nog wel met een pre-release die geprobeerd kan worden, maar ik denk dat de huidige staat in de kern een goede basis is. 
Ik heb nog wel veel meer grafische zaken die ik wil proberen, maar laat ik maar niet teveel in 1x doorvoeren... :]

dennissiemensma, 2021-04-03:
De nieuwe opzet is nu onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).

dennissiemensma, 2021-04-03:
@donnie-dominguez @cmderks93 jullie kunnen weer terugwisselen met:

```
sudo su - dsmr
git checkout v4
./deploy.sh
```

Mochten jullie dat inmiddels nog niet gedaan hebben.

cmderks93, 2021-04-03:
Top! Ga de boel updaten!

donnie-dominguez, 2021-04-06:
Ook m'n 'productie' machine draait nu de nieuwe versie. En het probleem met de kleuren van de teruglevering is ook hier netjes opgelost zoals hieronder is te zien. Top!

[image]

donnie-dominguez, 2021-04-06:
[image]
[image]
Ik weet niet of het bewust is, maar in de pop-up staan de kleuren bij het hooveren op de levering anders  (eerst de Lx- en dan Lx+) dan bij het hooveren over de grafiek van terugleveren.

dennissiemensma, 2021-04-06:
@donnie-dominguez bedankt voor de input. Dat viel mij destijds ook al op, maar ik kreeg het niet gefixt in de pop-up. Wellicht dat ik er ooit nogmaals naar kijk.