dsmrreader/dsmr-reader #999: Onverwerkte telegrammen lopen op

rodegra-rdg, 2020-05-25:
Paar weken geleden heb ik alles opnieuw geïnstalleerd en een backup terug gezet.
Allemaal prima gegaan.
Tot ik van de week zag dat mijn live grafieken niets meer lieten zien.
Op de status pagina zag ik heel veel onverwerkte telegrammen en dat de verwerking bijna 2 dagen achter liep.
vanmiddag een backup van de 11e terug gezet.
In eerste instantie zag ik de onverwerkte berichten weer verminderen en ook de live grafieken gaven weer beeld. nu zie ik helaas dat de berichten weer gaan achter lopen en de tijd van de live grafieken ook.
Enig idee wat ik hier aan zou kunnen doen?

Bedankt Alvast

dennissiemensma, 2020-05-25:
Bedankt voor je melding. Kun je aangeven welke versie je draait, welke DSMR-versie je slimme meter heeft en waar je het op draait?

Het kan te maken hebben met een te grote dataset, maar zonder meer informatie blijft het gissen voor mij.

Daarnaast kun je debug-logging inschakelen om te zien of de verwerking ergens tegenaan loopt: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

rodegra-rdg, 2020-05-25:
vergeten te vermelden sorry.
Het gaat om versie 3.9.1, alles staat op een Raspberry Pi 3+ en slimme meter SMR 5.0
Zie nu net dat de berichten langzaam weer aan het afnemen zijn.

rodegra-rdg, 2020-05-25:
In dsmrreader.log kom ik veel van deze meldingen tegen.
Ik weet niet in welke log ik verder nog wat kan halen, dubug staat aan.

dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '56632' (DD38) does not match the telegram checksum '22529' (5801)
[2020-05-25 20:01:14,667] WARNING  @ datalogger | Rejected telegram (Invalid telegram CRC. The calculated checksum '40275' (9D53) does not match the telegram checksum '32756' (7FF4)) (base64 encod$
[2020-05-25 20:01:14,668] ERROR    @ datalogger | Invalid telegram CRC. The calculated checksum '40275' (9D53) does not match the telegram checksum '32756' (7FF4)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 92, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 42, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 88, in validate_checksum
    expected_crc_hex
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '40275' (9D53) does not match the telegram checksum '32756' (7FF4)
[2020-05-25 20:04:41,898] WARNING  @ datalogger | Rejected telegram (Invalid telegram CRC. The calculated checksum '27287' (6A97) does not match the telegram checksum '63824' (F950)) (base64 encod$
[2020-05-25 20:04:41,900] ERROR    @ datalogger | Invalid telegram CRC. The calculated checksum '27287' (6A97) does not match the telegram checksum '63824' (F950)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 92, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 42, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 88, in validate_checksum
    expected_crc_hex
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '27287' (6A97) does not match the telegram checksum '63824' (F950)

rodegra-rdg, 2020-05-26:
teller staat nu op 99 achterlopende telegrammen, lijkt nu wel aardig te lopen.
Voor ik gisteren de backup had terug gezet, stond de teller op ruim 139000 en een achterstand van 2 dagen.
Ik hou het nog even in de gaten.

dennissiemensma, 2020-05-26:
Die foutmeldingen in de dsmrreader log mag je negeren als je verder wel af en toe een goed telegram binnenkreeg.

Het ging vooral om de dsmr-backend log waar debug-info in staat.

Als de achterstand nu ingelopen is en alles weer werkt, vergeet vooral niet om de debug-log uit te zetten. Gezien het je SD-kaartje sneller doet slijten door de vele writes.

rodegra-rdg, 2020-05-26:
In de dsmt-backend.log zie ik eigenlijk geen bijzonderheden, debug inmiddels uitgezet.
Tot nu toe zie ik de teller rond de 100 staan.
Vooralsnog lijkt het dus goed te gaan.
Wat ik wel merk is dat als ik naar de live grafieken ga, dit even bekijk en dan weer terug ga naar de status pagina de onverwerkte telegrammen met ruim 100 is opgelopen.

dennissiemensma, 2020-05-27:
Je kunt overwegen om de 'sleep' van de datalogger te verhogen. Dit kan in de admin-interface. 

Als je die op 5 of 10 seconden zet, beperkt dat de doorvoer van de telegrammen ontzettend (voor DSMR5-meters), zonder dat er verschil is in je dag- of uurtotalen.

Daarnaast kan het helpen om retentie anders in te stellen, zodat oude metingen na een periode worden opgeschoond. De dagtotalen blijven altijd bewaard.

rodegra-rdg, 2020-05-28:
Ik heb beide tips toegepast. Voor nu ga ik het even in de gaten houden.
Dank voor je medewerking.

closed 2020-05-28 by dennissiemensma