dsmrreader/dsmr-reader #1003: dsmr_mqtt BACKOFF   unknown error making dispatchers for 'dsmr_mqtt': ENOSPC

MORD0R, 2020-05-27:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie 3.9.1
*  RaspberryPi ...
* Native 
* DSMR-protocol slimme meter *(v4 of v5)*: v

[image]

[image]

Deze instructie gevolgd
https://dsmr-reader.readthedocs.io/nl/v3/mqtt.html

dennissiemensma, 2020-05-27:
Bedankt voor je melding. Helaas heb ik deze fout nog niet eerder gezien.

Wat staat er in `/var/log/supervisor/dsmr_mqtt.log`?

MORD0R, 2020-05-28:
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
[2020-01-27 23:00:10,146] WARNING  MQTT | --- Detected settings change, requiring process restart, stopping...
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
[2020-01-28 19:52:10,176] WARNING  MQTT: --- Unexpected disconnect, re-connecting...
[2020-01-28 19:52:10,204] ERROR    MQTT: Failed to re-connect to broker: [Errno 111] Connection refused
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 195, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?
could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 46, in handle
    self.data = self.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/management/commands/dsmr_mqtt.py", line 23, in initialize
    self.sleep_time = MQTTBrokerSettings.get_solo().process_sleep
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 538, in get_or_create
    return self.get(**kwargs), False
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 402, in get
    num = len(clone)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 256, in __len__
    self._fetch_all()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1242, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 55, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1131, in execute_sql
    cursor = self.connection.cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 256, in cursor
    return self._cursor()

MORD0R, 2020-05-28:
Dit is vreemd een herstart gedaan en het werkt. (had ik al meerdere keren gedaan)

Ik ben benieuwd voor hoelang het blijft werken.

dennissiemensma, 2020-05-28:
Bedankt voor het delen. De fouten geven aan dat de database-verbinding in DSMR-reader weggevallen is. Als dat regelmatig blijft voorkomen, dan zou je nog wat dieper kunnen zoeken in de logs.

MORD0R, 2020-05-28:
Ik heb ook een paar twee keer het pid file moeten verwijderen sinds ik dmsr reader gebruikt anders starte hij echt niet meer op. (na een stroom uitval) Misschien heeft dit een relatie? Ik draai op deze pi ook openhab en zigbee2mqtt en deze twee blijven wel altijd werken. Ik gebruik alleen de monitor op deze pi.

dennissiemensma, 2020-05-28:
Dat zou kunnen. Ging het altijd over dsmr_mqtt of een ander proces dat niet wilde starten door de pidfiles?

MORD0R, 2020-05-29:
de webinterface.

dennissiemensma, 2020-06-24:
@MORD0R hoe is de situatie inmiddels, komt dit nog steeds bij je voor?

MORD0R, 2020-06-29:
lijkt zich niet meer voor te doen.

dennissiemensma, 2020-06-29:
Oke dank voor je terugkoppeling! Dan laat ik het er voor nu even bij.

closed 2020-06-29 by dennissiemensma