dsmrreader/dsmr-reader #1004: PID file does not exist

vincedh, 2020-05-27:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.9.1
* Hardware *(RaspberryPi 3, 4 of iets anders)*: esxi + ubuntu server
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: native
* DSMR-protocol slimme meter *(v4 of v5)*: v.5

Hoi Dennis,
Ik heb een nieuwe installatie gedaan van DSMR reader en een backup teruggezet, een Pi stuurt berichten vanaf de meterkast via de API naar DSMR. Tot zover gaat alles goed.
Alleen mn MQTT berichten blijven hangen ( Wachtende berichten: 46)

Als ik  ./post-deploy.sh doe krijg ik onderaan:
` * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist
`
In dsmr_datalogger.log staat alleen het volgende:
`The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging`

Heb jij enig idee?

dennissiemensma, 2020-05-27:
Standaard wordt het proces voor MQTT niet (meer) geinstalleerd. Dit proces verstuurt alle uitgaande berichten. In DSMR-reader v4.x komt het proces er (weer) standaard in. 

Voor nu is het nog even handwerk voor je: https://dsmr-reader.readthedocs.io/nl/v3/mqtt.html#recent-versions-v3-0-0-or-higher

dennissiemensma, 2020-05-27:
Los van MQTT: Je kunt de datalogger trouwens weghalen als DSMR-reader niet meer direct aan de meter zit:

```
    sudo rm /etc/supervisor/conf.d/dsmr_datalogger.conf
    sudo supervisorctl reread
    sudo supervisorctl update
```

vincedh, 2020-05-27:
Ontzettend bedankt! Werkt weer als een speer!
Datalogger ook weggehaald, nogmaals hartelijk dank.

closed 2020-05-27 by dennissiemensma