dsmrreader/dsmr-reader #1007: MQTT: interval instellen

Thuurke, 2020-05-30:
Is het mogelijk om een interval in te bouwen bij MQTT?
Ik heb mqtt nu gekoppeld aan Home Assistant DSMR reader component en volgens de logging worden berichten nu voor sommige topics zelfs x keer per seconden gestuurd. Ik denk dat het handig is al je een interval zou kunnen instellen, zodat de topics bv iedere x seconden gepublished worden.

Thuurke, 2020-05-30:
Ok het zit er blijkbaar al in.. (beperkt tot 2 cijfers voor de komma) admin/dsmr_mqtt/mqttbrokersettings/ en dan MQTT proces sleep:
Al staat het niet in de documentatie https://dsmr-reader.readthedocs.io/nl/v3/admin/mqtt.html genoemd of afgebeeld.

dennissiemensma, 2020-05-31:
Bedankt voor je vraag. Het idee van MQTT is dat meldingen of events realtime zijn en daarom worden ze niet vertraagd.

Wat je wel kunt doen is de bron vertragen, bijvoorbeeld de sleep van de datalogger of backend. De MQTT sleep kun je, zoals je aangeeft, ook gebruiken. Maar die is van origine bedoeld voor de performance bij problemen. :]

closed 2020-05-31 by dennissiemensma