dsmrreader/dsmr-reader #1008: After upgrading to 3.10 (latest) no new telegrams are processed

pimvullers, 2020-06-04:
- DSMR-reader versie:  3.10
- Type RaspberryPi or server: QNAP NAS, Container station (i.e. docker)
- Standaardinstallatie of Docker: Docker - https://github.com/xirixiz/dsmr-reader-docker
- DSMR-protocol slimme meter v4/v5: 4.2 (Landis Gyr+ E350)

---

After updating to latest docker image (3.10), no new telegrams are processed.
I first expected it had to do with QNAP firmware upgrade that I combined (and might have messed with the permissions or so), but when entering the running docker, and listening on ttyUSB0 with microcom, I see telegrams. So the telegrams are accessable from withing the container, they are only not recorded/processed by DSMR reader.
After a hint from xirixiz I checked the logs, and it looks like the CRC of the telegrams are no longer accepted :(
I tried disconnecting and re-connecting the USB cable, rebooting the device, but no luck.

Log is full with messages like this:
```
[2020-06-04 10:52:37,469] WARNING  @ datalogger | Rejected telegram (Invalid telegram CRC. The calculated checksum '47633' (BA11) does not match the telegram checksum '26955' (694B)) (base64 encoded): b'L1hNWDVMR0JCRkcxMDEyNzQ0OTQ3DQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgyM
DA2MDQxMDUxMzlTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzMwMzAzMDMzMzczMjM2MzQzMTMzMzAzMTM2KQ0KMS0wOjEuOC4xKDAwOTY0NS4zNzAqa1doKQ0KMS0wOjEuOC4yKDAwNzQ2MC43MDMqa1doKQ0KMS0wOjIuOC4xKDAwMDAwMC4wMDAqa1doKQ0KMS0wOjIuOC4yKDAwMDAwMC4wMDAqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMikNCjEtMDoxLjcuMCgwMC41O
TIqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMDIpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAxKQ0KMS0wOjcyLjMyLjAoMDAwMDEpDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyL
jM2LjAoMDAwMDApDQowLTA6OTYuMTMuMSgpDQowLTA6OTYuMTMuMCgpDQoxLTA6MzEuNy4wKDAwMipBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjMxNiprVykNCjEtMDo0MS43LjAoMDAuMjYzKmtXKQ0KMS0wOjYxLjcuMCgwMC4wMTEqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoM
DAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzEzOTMzMzQzMDMzMzIzODMwMzgzOTM5MzEzNikNCjAtMToyNC4yLjEoMjAwNjA0MTAwMDAwUykoMDY1NTQuMjM5KjMpDQohNjk0Qg0K'                                                                            
[2020-06-04 10:52:37,469] ERROR    @ datalogger | Invalid telegram CRC. The calculated checksum '47633' (BA11) does not match the telegram checksum '26955' (694B)                                                                                                              
Traceback (most recent call last):                                                                                                                                                                                                                                              
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 92, in telegram_to_reading                                                                                                                                                                                          
    parsed_telegram = parser.parse(data)                                                                                                                                                                                                                                        
  File "/dsmr/dsmr_parser/parsers.py", line 42, in parse                                                                                                                                                                                                                        
    self.validate_checksum(telegram_data)                                                                                                                                                                                                                                       
  File "/dsmr/dsmr_parser/parsers.py", line 82, in validate_checksum                                                                                                                                                                                                            
    raise InvalidChecksumError(                                                                                                                                                                                                                                                 
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '47633' (BA11) does not match the telegram checksum '26955' (694B)                                                                                                                   
[2020-06-04 10:52:47,491] WARNING  @ datalogger | Rejected telegram (Invalid telegram CRC. The calculated checksum '23303' (5B07) does not match the telegram checksum '42795' (A72B)) (base64 encoded): b'L1hNWDVMR0JCRkcxMDEyNzQ0OTQ3DQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgyM
DA2MDQxMDUxNDlTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzMwMzAzMDMzMzczMjM2MzQzMTMzMzAzMTM2KQ0KMS0wOjEuOC4xKDAwOTY0NS4zNzAqa1doKQ0KMS0wOjEuOC4yKDAwNzQ2MC43MDUqa1doKQ0KMS0wOjIuOC4xKDAwMDAwMC4wMDAqa1doKQ0KMS0wOjIuOC4yKDAwMDAwMC4wMDAqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMikNCjEtMDoxLjcuMCgwMC42M
jkqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMDIpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAxKQ0KMS0wOjcyLjMyLjAoMDAwMDEpDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyL
jM2LjAoMDAwMDApDQowLTA6OTYuMTMuMSgpDQowLTA6OTYuMTMuMCgpDQoxLTA6MzEuNy4wKDAwMipBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjMyOSprVykNCjEtMDo0MS43LjAoMDAuMjg3KmtXKQ0KMS0wOjYxLjcuMCgwMC4wMTEqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoM
DAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzEzOTMzMzQzMDMzMzIzODMwMzgzOTM5MzEzNikNCjAtMToyNC4yLjEoMjAwNjA0MTAwMDAwUykoMDY1NTQuMjM5KjMpDQohQTcyQg0K'                                                                            
[2020-06-04 10:52:47,491] ERROR    @ datalogger | Invalid telegram CRC. The calculated checksum '23303' (5B07) does not match the telegram checksum '42795' (A72B)                                                                                                              
Traceback (most recent call last):                                                                                                                                                                                                                                              
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 92, in telegram_to_reading                                                                                                                                                                                          
    parsed_telegram = parser.parse(data)                                                                                                                                                                                                                                        
  File "/dsmr/dsmr_parser/parsers.py", line 42, in parse                                                                                                                                                                                                                        
    self.validate_checksum(telegram_data)                                                                                                                                                                                                                                       
  File "/dsmr/dsmr_parser/parsers.py", line 82, in validate_checksum                                                                                                                                                                                                            
    raise InvalidChecksumError(                                                                                                                                                                                                                                                 
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '23303' (5B07) does not match the telegram checksum '42795' (A72B) 
```

Example of manually received telegram:
```
[... cut ...]
0-1:24.2.1(200604110000S)(06554.239*m3)                                                                                                                                                                                                                                         
!0668                                                                                                                                                                                                                                                                           
```

Any ideas what could cause this sudden CRC corruption for all new datagrams?

dennissiemensma, 2020-06-04:
Bedankt voor je melding. Ik heb dit niet eerder gezien. 

Als de output van je onderste telegram exact is zoals je hem ontvangen hebt, dan lijken er inderdaad wat tekens te missen.

Verder heb ik even een van de base64 strings gedecode (omdat er ook `\r\n` in staat) en hier geprobeerd. Eentje lijkt dit te bevatten:
```
/XMX5LGBBFG1012744947\r
\r
1-3:0.2.8(42)\r
0-0:1.0.0(200604105139S)\r
0-0:96.1.1(4530303330303033373236343133303136)\r
1-0:1.8.1(009645.370*kWh)\r
1-0:1.8.2(007460.703*kWh)\r
1-0:2.8.1(000000.000*kWh)\r
1-0:2.8.2(000000.000*kWh)\r
0-0:96.14.0(0002)\r
1-0:1.7.0(00.592*kW)\r
1-0:2.7.0(00.000*kW)\r
0-0:96.7.21(00002)\r
0-0:96.7.9(00000)\r
1-0:99.97.0(0)(0-0:96.7.19)\r
1-0:32.32.0(00000)\r
1-0:52.32.0(00001)\r
1-0:72.32.0(00001)\r
1-0:32.36.0(00000)\r
1-0:52.36.0(00000)\r
1-0:72.36.0(00000)\r
0-0:96.13.1()\r
0-0:96.13.0()\r
1-0:31.7.0(002*A)\r
1-0:51.7.0(002*A)\r
1-0:71.7.0(000*A)\r
1-0:21.7.0(00.316*kW)\r
1-0:41.7.0(00.263*kW)\r
1-0:61.7.0(00.011*kW)\r
1-0:22.7.0(00.000*kW)\r
1-0:42.7.0(00.000*kW)\r
1-0:62.7.0(00.000*kW)\r
0-1:24.1.0(003)\r
0-1:96.1.0(4730303139333430333238303839393136)\r
0-1:24.2.1(200604100000S)(06554.239*3)\r
!694B\r
```

Wanneer ik die in een test probeer, faalt de CRC check inderdaad. 

Het telegram lijkt overigens verder op het oog goed, op 1 byte na in de gasstand:
```
0-1:24.2.1(200604100000S)(06554.239*3)\r
```

Als ik er dit van maak (`3` wordt `m3`):
```
0-1:24.2.1(200604100000S)(06554.239*m3)\r
```

Dan gaat de CRC check goed.

Dus het lijkt er inderdaad op dat de telegrammen niet meer goed binnenkomen. Het laatste telegram in je eigen comment mist dus sowieso veel bytes op het oog, er vanuitgaande dat je het goed hebt gekopieerd uit je terminal.

Ik zou alleen niet weten hoe dit verder op te lossen helaas. Ik neem aan dat alles gewoon werkte tot het moment dat je de update/wijziging had uitgevoerd?

Je kunt ook nog de vorige versie van DSMR-reader proberen, maar als de telegrammen in `cu` al incorrect binnenkomen, verandert DSMR-reader qua versie er natuurlijk niets aan.

Je kunt je nog situatie qua kabel/uitlezen en CRC voorleggen in een groot topic als: https://gathering.tweakers.net/forum/list_message/62534590

Mogelijk heeft iemand anders hetzelfde meegemaakt.

pimvullers, 2020-06-08:
Ik heb vandaag nog eens wat geprobeerd. Kabel uit de meter, en weer erin, geen effect.
USB kabel eruit, bij andere poort erin, en nu doet hij het weer. 
Het lijkt dus een lokaal probleem.

Dank voor dit mooie software pakket.

closed 2020-06-08 by pimvullers