dsmrreader/dsmr-reader #1009: Prijzen opnieuw berekenen voor teruglevering.

Mixpower, 2020-06-04:
Doordat ik tarieven heb aangepast moest ik de prijzen opnieuw berekenen,
maar na dat gedaan te hebben zijn de prijzen voor teruglevering compleet verdwenen.

De prijzen voor het verbruik kloppen wel gewoon maar de teruglevering is dus na opnieuw berekenen helemaal weg, is hier een oplossing voor?

Bedankt alvast!

dennissiemensma, 2020-06-04:
Bedankt voor je melding. Het is mij niet helemaal duidelijk welke prijzen verdwenen zijn. 

- Bedoel je de tarieven? Of de dagtotalen? 
- Wellicht kun je een screenshot of iets maken?
- En heb je deze uitgevoerd voor de herberekening: `dsmr_stats_recalculate_prices`?

Mixpower, 2020-06-04:
De teruglevering word niet meer mee berekend in kosten, dus de bedragen zijn nu veel hoger en kloppen dus niet meer omdat de teruglevering er niet meer afgetrokken is na een herberekening.

Dit geld zowel voor archief als de Live grafieken.

Ja ik heb `dsmr_stats_recalculate_prices` gebruikt.

dennissiemensma, 2020-06-06:
Bedankt voor je aanvulling. Ik heb het inmiddels kunnen vinden en oplossen voor de volgende release. 

Je kunt dan tzt `dsmr_stats_recalculate_prices` simpelweg nogmaals uitvoeren.

closed 2020-06-06 by dennissiemensma

dennissiemensma, 2020-06-17:
Het is gereleased, dus je kunt het na de update nogmaals proberen.

Mixpower, 2020-06-17:
> Het is gereleased, dus je kunt het na de update nogmaals proberen.

Net getest en de prijzen kloppen nu weer, thanks voor het fixen!