dsmrreader/dsmr-reader #1011: API mislukt Authorization Required

Sjaak1982, 2020-06-04:
Hoi Dennis,

Ik probeer de API lokaal op de Raspberry 3. 
Krijg alleen maar de foutcode 401  Authorization Required

Heb de API-call toegestaan in de configuratie met als API "Test"

Dit is mijn python script:

```
import requests
import json

response = requests.get(
    'http://localhost/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1',
    headers={'X-AUTHKEY' : 'Test'},
)

if response.status_code != 200:
    print('Error: {}'.format(response.text))
else:
    print('Response: {}'.format(json.loads(response.text)))
```

Weet jij wat er fout kan zijn.
Kan ik nog ergens in een logfile bekijken waarom ik error 401 krijg?

dennissiemensma, 2020-06-04:
Je kunt sowieso even overwegen om Postman te installeren en daarin de kant-en-klare collectie + variabelen in te vullen. Deze vind je in DSMR-reader op de URL `/docs/api#section/Postman-examples`.
Als je geen ervaring hebt met Postman, na het importeren van de collectie en env kun je de env selecteren en vul je daar je eigen URL + API key in. Voorbeeld: https://learning.postman.com/docs/postman/variables-and-environments/managing-environments/#selecting-an-active-environment

Verder kun je even kijken naar de remote datalogger: https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py#L90

Die gebruikt een wat recentere header met `headers={'Authorization': 'Token Test'}`.

Sjaak1982, 2020-06-05:
Het is de bedoeling om het via een script lokaal op de raspberry te doen waar ook DSMRreader staat.
Postman is dan geen oplossing. 
Ook met curl lukt het niet.

```
curl 'http://localhost/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1' \
    -H 'X-AUTHKEY: Test' 
```
Resultaat:
```
[<html>
<head><title>401 Authorization Required</title></head>
<body bgcolor="white">
<center><h1>401 Authorization Required</h1></center>
<hr><center>nginx/1.10.3</center>
</body>
</html>]
```

dennissiemensma, 2020-06-06:
Ik begrijp dat je het lokaal wilt doen, maar ik had even moeten vermelden dat mijn suggestie bedoeld was om eerst te kijken of je uberhaupt via Postman bij de API kan. Als dat werkt, is het stukken makkelijker te debuggen. Het is niet bedoeld als oplossing.

Verder kan ik je dus aanraden om het ook eens te proberen met een andere auth header zoals beschreven in mijn vorige comment en het voorbeeldscript.

dennissiemensma, 2020-06-11:
@Sjaak1982 ben je hier nog verder mee gekomen?

Sjaak1982, 2020-06-15:
Ik heb het geprobeerd, maar krijg ook weer de 401 error. Zie geen andere meldingen.

dennissiemensma, 2020-06-15:
Bedankt voor je update. Kan het zijn dat je er nog een HTTP Basic Auth voor hebt gezet? 

Bijvoorbeeld voor als je je installatie ook publiekelijk toegankelijk hebt gemaakt en er een user + wachtwoord op hebt gezet.

Als ik namelijk via Postman test, krijg ik bij foute authenticatie altijd een ``HTTP 403``. Als ik er Basic Auth voor zet en dat probeer zonder goede login, krijg ik ook een ``HTTP 401``.

Dat zou tevens verklaren waarom de API key niet lijkt te werken bij jou.

Sjaak1982, 2020-06-16:
Probleem is gevonden. 
Ik heb de configuratie van nginx nog eens gecontroleerd.
In het bestand /etc/nginx/sites-enabled/dsmr-webinterface stond achter server allow een cijfer fout in de ip-range
allow 192.168.17**7**.0/24     gewijzigd in allow 192.168.17**8**.0/24
daarna nginx opnieuw gestart en API calls werken nu. 
Heb nog wel een timeout bij /api/v2/datalogger/dsmrreading maar dat komt doordat er meer dan 7 miljoen records in de tabel dsmr_datalogger_dsmrreading staan. Ik heb een v5 meter die elke seconde een datagram geeft.
Datalogger proces sleep staat nu op 10.0 en Dataretentie stond op 6 maanden en heb daar 3 maanden van gemaakt. 
(Zelfde probleem als in issue #973 is gemeld)

dennissiemensma, 2020-06-16:
Bedankt voor je terugkoppeling en fijn dat je het hebt kunnen vinden!

Wat betreft de performance van de API, heb je een specifieke call die traag is? Is dat dezelfde op: `/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1`?

Op een Pi zal de performance (bij grote datasets) altijd wat minder zijn dan op een echte server of PC, maar het kan natuurlijk ook zijn dat ik of je lokale installatie nog een DB index mist die het oplost.

Sjaak1982, 2020-06-16:
Het is inderdaad de API dsmrreading. Doet het nu wel, maar heeft 30 seconden nodig.
Er staan nu 5 miljoen records in de tabel.
Ik heb ook vacuumdb uitgevoerd.

Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_processed_add8a852" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" btree ("timestamp")

dennissiemensma, 2020-06-16:
Welke parameters geef je mee aan de API-call? Wat probeer je op te halen? Alleen de meest recente dsmrreading?

Sjaak1982, 2020-06-16:
het laatste record:
URL = 'http://localhost/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1'

dennissiemensma, 2020-06-17:
Bedankt voor de terugkoppeling. Ik kan die helaas niet sneller krijgen nu.

Mocht je overigens op zoek zijn naar iets wat simpelweg elk binnengekomen telegram doorstuurt naar iets, dan kun je als alternatief ook nog overwegen om een plugin te maken: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-4-forwarding-dsmr-readings-in-json-format-to-some-api

closed 2020-06-17 by dennissiemensma

Sjaak1982, 2020-06-17:
Met de volgende query is het laatste record binnen een seconde gelezen. 

`psql -U postgres -d dsmrreader -c 'SELECT * from dsmr_datalogger_dsmrreading order by timestamp desc limit 1'`

Omdat het lokaal wordt gebruikt, ga ik bovenstaande query in een script verwerken.
Bedankt voor de support.