dsmrreader/dsmr-reader #1012: New energysupplier contract not picked up if older one still active

pa1pdr, 2020-06-06:
I added new supplier prices, which DSMR does not pick up and ignores when an older contract is still active. Expected behaviour would be that it takes the most recent start date until the end-date, but with below table no supplier price is picked up and cost are depicted as -.

Only when the second cord is amended to have an enddate as 31-5, DSMR picks up the supplier prices. 

postgresql table
` id |   start    |    end     | electricity_delivered_1_price | electricity_delivered_2_price | gas_price |     description     | electricity_returned_1_price | electricity_returned_2_price 
----+------------+------------+-------------------------------+-------------------------------+-----------+---------------------+------------------------------+------------------------------
  1 | 2018-09-01 | 2019-09-03 |                       0.20121 |                       0.21721 |   0.75009 | Energie Direct Flex |                      0.00000 |                      0.00000
  2 | 2019-09-04 | 2020-09-01 |                       0.21521 |                       0.23121 |   0.75909 | Energie Direct flex |                      0.00000 |                      0.00000
  3 | 2020-06-01 | 2021-06-01 |                       0.20537 |                       0.21749 |   0.68979 | Van de Bron         |                      0.00000 |                      0.00000`

dennissiemensma, 2020-06-06:
Thanks for reporting your issue. I understand that managing the contracts is not ideal, as it lacks some checks in overlapping dates. This is because the project uses the builtin admin feature of Django, which lacks some flexibility for managing such constraints.

That being said, you should not expect the behavior to be as described, as it could also be expected to work the other way around. 
Therefor when DSMR-reader detects a conflict in contract dates, it just ignores all prices for it, preventing it to decide for itself which one should be used, causing to use faulty prices. Which may be hard to detect, since the differences between contracts tend to be slim. 

For now you'll have to make sure the contracts do not overlap and you can fix the prices later with: https://dsmr-reader.readthedocs.io/en/latest/faq.html#recalculate-prices-retroactively

dennissiemensma, 2020-08-03:
I was able to add some additional validation for this situation in the admin. Will be released along v4.2

closed 2020-08-03 by dennissiemensma

dennissiemensma, 2020-08-19:
Uitgebracht in [v4.2](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.2.0)

pa1pdr, 2020-08-19:
Thanks for picking this up, it was not material but good to have it fixed 👍 🥇