dsmrreader/dsmr-reader #1013: Pip install psycopg2 vervangen door OS package

dennissiemensma, 2020-06-06:
Onderzoeken of het mogelijk is om `python3-psycopg2` te gebruiken als alternatief. Dit scheelt weer een dependency die zelf gecompiled moet worden.

rippiedoos, 2020-06-06:
Door de package python3-psycopg2 te installeren en de virtualenv aan te maken met --system-site-packages ipv --no-site-packages hoeven alle development-spullen er niet opgezet te worden. Zaken zoals gcc, g++, cpp en verschillende dev-packages (zoals postgresql-dev) zijn niet meer nodig. Ik heb dit met getest op Ubuntu 18.04 en ik heb de volgende packages ook meegenomen vanuit het OS, die heel erg dicht of helemaal in de buurt kwamen van de verwachte packages in base.txt:

    python3-serial
    python3-coreapi
    python3-crcmod

In relatie tot site-packages is tegenwoord --system-site-packages de standaard. --no-site-packages is tegenwoordig deprecated en zou niet meer gebruikt moeten worden. En het scheelt een hoop gecompile door de psycopg2 vanuit het OS mee te nemen! De overige pip-packages zijn allemaal native python, dus daar hoeft weinig meer voor te gebeuren dan downloaden.

closed 2020-06-11 by dennissiemensma

dennissiemensma, 2020-06-24:
@rippiedoos ik kom nog wel een interessante situatie tegen voor Ubuntu 16.4. Daar is Python3.5 nog standaard, maar DSMR-reader is Python 3.6.

Als ik dan `sudo apt-get install python3-psycopg2` doe dan pakt die de `3.5` versie en in de virtualenv lijkt dat dan mis te gaan.

Gezien deze LTS nog maar een jaar ondersteund wordt en de volgende LTS standaard Python `3.6` heeft, lijkt het me geen groot issue. In het ergste geval wijs ik gebruikers wel op het installeren en compilen via `pip`, zoals het was.

rippiedoos, 2020-06-25:
Op Ubuntu 16.04 is dat wel een probleem dan ja, dan zou psycopg2-binary weer een betere oplossing zijn. Moet je nog wel de libpq-library als requirement erbij zetten (vermoed ik). Ik weet niet of libpq standaard meekomt als package, maar dat zal wel denk ik. Op CentOS kun je die los installeren namelijk.

dennissiemensma, 2020-08-17:
@rippiedoos ik merk toch steeds meer en meer de nadelen van `--system-site-packages`. 

Ik kom issues tegen zoals:
- dennissiemensma/dsmr-reader/issues/1078 waarbij Gunicorn niet te gebruiken is op het hardcoded pad dat in Supervisor staat.
- lokaal bij mij werkte `pytest` niet meer door een of ander conflict met de systeemversie.

Ik zal daarom kijken of het later weer zonder kan en dan eventueel `psycopg2-binary` gebruiken als alternatief.

rippiedoos, 2020-08-18:
Waarom wordt dan ook supervisor gebruikt? Systemd gebruiken lijkt mij een stuk handiger. 

Lijkt erop dat er geknoei gaat komen OS-versies en dat het op een bepaalde manier niet meer gaat werken. Persoonlijk zou ik die hard coded paden dus niet gebruiken in zowel supervisor en/of systemd maar het PATH aanpassen zodat de virtualenv/bin vooraan staat.

dennissiemensma, 2020-08-18:
Bedankt voor je terugkoppeling. De reden dat er Supervisor wordt gebruikt is omdat het een eenvoudig en vooral robuust stukje software is wat zichzelf al lange tijd heeft bewezen en vrij standaard is.

Dat zeggende hebben, voor Gunicorn zijn er [inmiddels meerdere mogelijkheden](https://docs.gunicorn.org/en/latest/deploy.html), waar Systemd ook als alternatief staat. Echter, Supervisor is zoveel eenvoudiger te beheren dan systemd, maar ik wil er nog wel een keer opnieuw naar kijken. 

En je suggestie voor het path is nog wel een goede, die zal ik ook nog proberen.