dsmrreader/dsmr-reader #1014: FireBase Authentication for Public use

squadramunter, 2020-06-09:
Ik heb gehoord over FireBase authenticatie. Heeft u hier ook eens over nagedacht om dit te gebruiken voor public use cases? Ik was bezig met het ontwikkelen van een Android app met Webview alleen die HTTP Authentication werkt daarop niet zonder dat ik daar al teveel moeite voor moet insteken om erachter te komen hoe ik dit werkend krijg. Zijn er misschien andere manieren om de site te beschermen zonder dat ik Basic Authentication gebruikt op mijn server?

dennissiemensma, 2020-06-09:
Bedankt voor je vraag. Basic auth is standaard optie voor webservers, maar daar is DSMR-reader niet aan gebonden. Het wordt genoemd in de documentatie omdat het het meeste simpele is om op te zetten en te gebruiken.

Voor het afschermen kun je ook overwegen om een VPN te gebruiken waarmee je in je thuisnetwerk kan. Bijvoorbeeld OpenVPN. Daarmee verleg je de complexiteit en beveiliging naar de VPN.

Wellicht dat ik in de toekomst een optie maak om de gehele applicatie achter een login te zetten.

squadramunter, 2020-06-09:
Ik heb voor nu een tijdelijke woraround gevonden om dit toch werkend te krijgen. Mijn wachtwoord staat alleen wel in plain text in mijn Android Applicatie maar ik gebruik het toch enkel voor mijzelf.
`            @Override
            public void onReceivedHttpAuthRequest(WebView view, HttpAuthHandler handler, String host, String realm) {
                handler.proceed("username", "password");
            }`