dsmrreader/dsmr-reader #1021: Instabiele installatie van DSMR op Raspberry Pi 3B met opslag op USB

Bashtastic, 2020-06-14:
Wat gebruik je?
- DSMR-reader versie:   3.10
- Type RaspberryPi or server:   3B
- Standaardinstallatie of Docker:   Standaard installatie
- DSMR-protocol slimme meter v4/v5:   v5

---

Een paar weken geleden is mijn sd kaart gecrashed en heb ik besloten om compleet opnieuw te beginnen. Hierbij was mijn doel om dit keer de database op een usb stick op te slaan en om een backup van mijn meetgegevens van het afgelopen jaar hierin te herstellen. Het herstellen van de oude data is mij niet gelukt, dit bleef hangen op velden die niet herkend werden, misschien dat dit komt omdat ik eerder versie 2 van DSMR draaide. Geen enorm debakel vind ik, zolang ik maar wel nieuwe data kan opslaan. Maar ook dat gaat niet vlekkeloos. De helft van de tijd draait de webinterface niet en regelmatig krijg ik meldingen dat de telegram berichten 12 uur achterlopen, en mqtt is al helemaal uit de lucht. De usb stick([deze](https://www.amazon.de/dp/B00RORBNR6?ref_=pe_3044161_327547611_E_304_dt_1)) is een USB 3.0 stick en ik heb een test gedaan waarbij ik ongeveer een schrijfsnelheid van 10 mb/s haalde.  Ik heb geprobeerd om wijs te worden uit de logging, maar het lukt me niet om goede aanknopingspunten te vinden. Ik hoop daarom dat je me enkele pointers zou kunnen geven.

`log dsmr backend
[2020-06-14 11:04:39,612] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-14 11:04:40,623] DEBUG    SP: 1 backend service(s) ready to run
[2020-06-14 11:04:40,624] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-06-14 11:04:40,664] DEBUG    Compact: Waiting for newer readings before grouping data...
[2020-06-14 11:04:40,676] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-06-14 11:04:55.665611+02:00 (ETA 0:00:14.988793)
[2020-06-14 11:04:40,721] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-14 11:04:41,732] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-14 11:04:41,776] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-14 11:04:42,788] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-14 11:04:42,830] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

log datalogger
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging

log dsmr webinterface
[2020-06-14 05:25:57 +0100] [683] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (683)
[2020-06-14 05:25:57 +0100] [683] [INFO] Using worker: sync
[2020-06-14 05:25:57 +0100] [692] [INFO] Booting worker with pid: 692
[2020-06-14 05:35:44 +0100] [683] [CRITICAL] WORKER TIMEOUT (pid:692)
[2020-06-14 06:35:44 +0200] [692] [INFO] Worker exiting (pid: 692)
[2020-06-14 05:35:44 +0100] [714] [INFO] Booting worker with pid: 714
[2020-06-14 09:32:13 +0100] [683] [INFO] Handling signal: hup
[2020-06-14 09:32:13 +0100] [683] [INFO] Hang up: Master
[2020-06-14 10:32:13 +0200] [714] [INFO] Worker exiting (pid: 714)
[2020-06-14 09:32:13 +0100] [2289] [INFO] Booting worker with pid: 2289

log django:
[2020-06-14 10:08:24,891] WARNING  @ log | Not Found: /phpmyadmin2012/index.php
[2020-06-14 10:08:24,991] WARNING  @ log | Not Found: /phpmyadmin2013/index.php
[2020-06-14 10:08:25,092] WARNING  @ log | Not Found: /phpmyadmin2014/index.php
[2020-06-14 10:08:25,194] WARNING  @ log | Not Found: /phpmyadmin2015/index.php
[2020-06-14 10:08:25,297] WARNING  @ log | Not Found: /phpmyadmin2016/index.php
[2020-06-14 10:08:25,399] WARNING  @ log | Not Found: /phpmyadmin2017/index.php
[2020-06-14 10:08:25,501] WARNING  @ log | Not Found: /phpmyadmin2018/index.php
[2020-06-14 10:08:25,603] WARNING  @ log | Not Found: /phpmyadmin2019/index.php
[2020-06-14 10:08:25,705] WARNING  @ log | Not Found: /phpmyadmin2020/index.php
[2020-06-14 10:08:25,807] WARNING  @ log | Not Found: /index.php`

dennissiemensma, 2020-06-14:
Bedankt voor je melding, het is een hele reeks aan issues maar je kunt het volgende proberen:

### Herstellen oude data
Als je die data nog hebt kan ik wel even voor je kijken of ik er tenminste de dagtotalen voor je uit kan halen, zodat je die weer kan terugzetten.

### Webinterface en achterstand
Dit kan duidelijk op performance issues, maar dat is niet zeker. Aangezien je DSMR v5 hebt, kan ik je aanraden te beginnen met het limiteren van het aantal telegrammen per minuut. Standaard heb je er 60, maar aan 5 a 10 per minuut heb je ook genoeg.
Je kunt in de admin in de datalogger config (`/admin/dsmr_datalogger/dataloggersettings/1/change/`) de waarde van `Datalogger proces sleep` ophogen. Deze kun je gerust naar 5 of 10 seconden zetten. Het kan een behoorlijke boost in de performance geven.

### MQTT
Als dit niet gerelateerd is aan het vorige punt, kan het zijn dat je die nog na (her)installatie handmatig moet inschakelen: https://dsmr-reader.readthedocs.io/nl/v3/mqtt.html

dennissiemensma, 2020-06-24:
Laat maar weten als je hier alsnog verdere hulp bij nodig hebt.

closed 2020-06-24 by dennissiemensma

Bashtastic, 2020-07-01:
Bedankt voor je hulp. Sorry dat ik wat langzaam ben in het reageren. Ik heb het nog eens overnieuw geïnstalleerd allemaal, maar het loopt nog steeds ongeveer 5 keer per dag vast, misschien is er gewoon wat mis met de raspberry pi. zal in de toekomst eens een andere setup proberen. MQTT werkt wel weer door je tip. Als DSMR vast loopt laat ik de pi opnieuw booten. Ik heb verder ingesteld dat de data ouder dan 7 dagen opgeschoond mag worden, misschien dat dat iets zal helpen. Het limiteren van het aantal telegrammen deed trouwens niks tov het aantal vastlopers per dag.

dennissiemensma, 2020-07-02:
Bedankt voor je update en jammer dat het het niet opgelost heeft. Aangezien je nu MQTT weer werkend hebt, kun je wellicht nog eens kijken in de logs of je iets ziet. Probeer ook vooral even of je iets geks ziet in de systeemlog `/var/log/syslog`.

Je zou ook nog eens kunnen kijken of je `atop` kan installeren: https://haydenjames.io/use-atop-linux-server-performance-analysis/
Deze tool maakt momentopnames van de proceslijst/belasting van je systeem op vaste intervallen en daar kun je dan na een vastloper achteraf naar kijken. Dit wordt namelijk gelogd op het bestandssyteem, want normaal kun je na een herstart niet meer zien welk proces veel CPU (of I/O) in beslag neemt.